pvlib/pvlib-python #2127: ModelChain Object doesn't accept Perez transposition model coefficents

bgpierc, 2024-07-11:
For comparison purposes, I'm running a lot of models using the `ModelChain` object.

 I'd like to vary the Perez parameters to see if my choice of parameters is having a significant effect, but using the `ModelChain` interface does not support changing this argument, which is passed to `system.get_irradiance` in `ModelChain.prepare_inputs`. 

This makes it so a `ModelChain` always uses the default `allsitescomposite1990` coefficients when using `perez`. 

We can work around this by using `ModelChain.run_model_from_poa` rather then `ModelChain.run_model`, but this seems rather clunky if all you want to change is the Perez coefficients. 

I'll submit a PR with a fix if there's interest. Thoughts?

cwhanse, 2024-07-11:
Do you have in mind adding a property like `ModelChain.transposition_model_kwargs` default to `None`?

bgpierc, 2024-07-11:
I think that's sensible, but might introduce some confusion since we have albedo as another argument in `pvsystem.get_total_irradiance`, but have it as an input in the weather data dataframe for the `ModelChain` object.  We'd want to avoid a `ModelChain.transposition_model_kwargs` that contains albedo, because then it'd be specified twice, I think. 

To avoid that, I just added a `model_perez` after `transposition_model`, which is ignored by `pvsystem.get_total_irradiance` if the `transposition_model` is not `perez`. I guess that could be confusing to people when printing the model, though.

cwhanse, 2024-07-11:
`pvsystem.get_total_irradiance already has `**kwargs` I was thinking that `ModelChain.transpositon_model_kwargs` would just be passed to that.

I am loathe to expand the `ModelChain` parameters, and adding `transposition_model_kwargs` invites other, similar additions.

What about adding a structure like `ModelChain.model_kwargs` that itself contains things like `transposition_model_kwargs`? That would keep the `ModelChain` API thin (adding only `model_kwargs=None`) while providing a vehicle for other additions.

I don't think it would need to contain all the parameters for `get_total_irradiance`, although it could and I get the possibility of a conflict with `weather`.

AdamRJensen, 2024-07-12:
> I'd like to vary the Perez parameters to see if my choice of parameters is having a significant effect, but using the `ModelChain` interface does not support changing this argument, which is passed to `system.get_irradiance` in `ModelChain.prepare_inputs`.

I think this is sufficiently niche that I would prefer burdening the user with creating the model chain from scratch rather than changing the API (fear of cluttering).

bgpierc, 2024-07-16:
Yeah, like I said, the workaround is quite simple so it's probably better to avoid the clutter. Maybe I can just put an example of that in the docs instead?

PR #2148 (2024-08-15): Perez coeff ModelChain example

 - [x] Closes #2127
 - [x] I am familiar with the [contributing guidelines](https://pvlib-python.readthedocs.io/en/latest/contributing.html)
 - [ ] Tests added
 - [ ] Updates entries in [`docs/sphinx/source/reference`](https://github.com/pvlib/pvlib-python/blob/main/docs/sphinx/source/reference) for API changes.
 - [x] Adds description and name entries in the appropriate "what's new" file in [`docs/sphinx/source/whatsnew`](https://github.com/pvlib/pvlib-python/tree/main/docs/sphinx/source/whatsnew) for all changes. Includes link to the GitHub Issue with `` :issu [... cut]
closed 2024-08-15 by kandersolar