pvlib/pvlib-python #2135: Create calculate_avg_photon_energy function

RDaxini, 2024-07-18:
I think a function to calculate the average photon energy (APE) using spectral irradiance data as an input would be a useful feature in pvlib.

**Additional context**
This would support PR #2126. While there are some discussions ongoing regarding that PR in Issue #2065, I think this may still be a useful feature in pvlib anyway.

The function would be in `pvlib/spectrum/spectral_irradiance.py` following the resolution of #2125 (I will create a PR for that soon issue soon)

I am wondering what people think about naming this function. Possibilities:

`calculate_avg_photon_energy`
`calculate_average_photon_energy`
`calculate_ape`
Switch "calculate" for "get", or just omit this first word entirely?
Alternatives? 

Any other thoughts on this?

echedey-ls, 2024-07-18:
> or just omit this first word entirely?

+1 for `average_photon_energy`

AdamRJensen, 2024-07-18:
> > or just omit this first word entirely?
> 
> +1 for `average_photon_energy`

+2

PR #2140 (2024-08-14): Create function to calculate average photon energy
- [X] Closes #2135 
 - [X] I am familiar with the [contributing guidelines](https://pvlib-python.readthedocs.io/en/latest/contributing.html)
 - [x] Tests added
 - [X] Updates entries in [`docs/sphinx/source/reference`](https://github.com/pvlib/pvlib-python/blob/main/docs/sphinx/source/reference) for API changes.
 - [x] Adds description and name entries in the appropriate "what's new" file in [`docs/sphinx/source/whatsnew`](https://github.com/pvlib/pvlib-python/tree/main/docs/sphinx/source/whatsnew) for all changes. Includes link to the GitHub  [... cut]
closed 2024-08-14 by kandersolar