pvlib/pvlib-python #2142: `CITATION.cff`: consider adding a file with the recommended and preferred citation entry

echedey-ls, 2024-07-22:
**Is your feature request related to a problem? Please describe.**
There is a recommended citation on the index page of the project, at https://pvlib-python.readthedocs.io/en/stable/#citing-pvlib-python
GitHub allows adding a file to the repo to show the recommended citation and make it easier to add it to a bibliography manager. Example of repo with that: https://github.com/pace577/logwright-solar-cell

In fact, there are three citations, so to make sure everybody reads each one of them in case there is one that suits the project the best, we may not want to add this file.

**Describe the solution you'd like**
I propose adding the recommended citation to this file.

**Describe alternatives you've considered**
- Take no action, so we make sure everybody goes through the index page and reads the other alternatives; and no future maintenance problems too (unless we link to the relevant files in the citation section of the webpage).
- Add a CITATION.bib with BibTeX comments (`% prefixed`) with explanations of which one to use
  https://docs.github.com/en/repositories/managing-your-repositorys-settings-and-features/customizing-your-repository/about-citation-files#other-citation-files
  GitHub would link to this file, but would not attempt to create an APA entry

**Additional context**
GitHub documentation on `CITATION.cff` (and other citation files): https://docs.github.com/en/repositories/managing-your-repositorys-settings-and-features/customizing-your-repository/about-citation-files

kandersolar, 2024-07-22:
Some previous discussion about CFF: #1322

echedey-ls, 2024-07-22:
Seems like I did not search well enough through the issues; closing as a duplicate.

Btw, most discussion was at #1388

closed 2024-07-22 by echedey-ls