pvlib/pvlib-python #2145: Fronius Symo Advanced 17.5-3-M missing

andreas-bulling, 2024-07-26:
the above inverter is missing in the database

https://www.fronius.com/de/solarenergie/installateure-partner/technische-daten/alle-produkte/wechselrichter/fronius-symo-advanced/symo-advanced-17-5-3-m

echedey-ls, 2024-07-26:
Do you mean this one? https://github.com/NREL/SAM/blob/1555ed5bdf259c2f373a61d34bd5033e37c6e2c9/deploy/libraries/CEC%20Inverters.csv#L562

Just so you know, pvlib-python only provides some databases distributed by the project above and some others. If the inverter is not there, you will need to wait until some lab characterizes it.

Two things:
- pvlib outdated databases do not contain any `"Symo"`
  <details><summary>Code</summary>
  <p>

  ```python
  from pvlib.pvsystem import retrieve_sam
  # %%
  d = retrieve_sam(name="CECMod")
  d.columns[d.columns.str.contains("Symo")]
  # %%
  d = retrieve_sam(name="SandiaMod")
  d.columns[d.columns.str.contains("Symo")]
  ```

  </p>
  </details> 
- Reading from the latest CEC Inverter database, first link on top of comment, gives an error
  ```python
  from pvlib.pvsystem import retrieve_sam
  retrieve_sam(path="https://github.com/NREL/SAM/blob/patch/deploy/libraries/CEC%20Inverters.csv")
  ```
  ```bash
  ParserError: Error tokenizing data. C error: Expected 1 fields in line 40, saw 18
  ```

Maintainers, have any of you came across this error? I read long time ago you modify the databases before distributing them through pvlib.

kandersolar, 2024-07-26:
Unfortunately, SAM will not be able to help with this either.  pvlib distributes the tables from SAM, but SAM is also merely distributing them from another source -- the California Energy Commission (CEC).  See here: https://www.energy.ca.gov/programs-and-topics/programs/solar-equipment-lists

This may also be helpful: https://pvlib-python.readthedocs.io/en/stable/user_guide/faq.html#the-cec-table-doesn-t-include-my-module-or-inverter-what-should-i-do

kandersolar, 2024-07-26:
@andreas-bulling you could also check older versions of the SAM database.  Sometimes older devices get removed from the database, so older versions may have the device you're looking for.  

In any case, this is not really a pvlib problem, so I will close this issue.  We already have #1345 to track the general issue of pvlib's tables being out of date.

closed 2024-07-26 by kandersolar

cwhanse, 2024-07-26:
@andreas-bulling I think the data you need are at the link you posted. I am not certain of the DC voltages used for the European efficiency values, but those look very similar to the CEC's measurement protocol.

andreas-bulling, 2024-07-26:
@cwhanse Thanks for your reply/help. I don't actually want to use the library directly. I want to use EMHASS (https://github.com/davidusb-geek/emhass-pvlib-database) but noticed that I can't because this inverter is missing. The EMHASS author recommended to open an issue here.

cwhanse, 2024-07-26:
> Maintainers, have any of you came across this error? I read long time ago you modify the databases before distributing them through pvlib.

I haven't. Sometimes the SAM team changes the format of that table between releases, and pvlib has to adapt.

This whole process (CEC tables -> model fitting -> SAM tables -> pvlib distributions) needs to be re-engineered for the benefit of both SAM and pvlib users. The SAM team has been open to cooperating.

0-k, 2024-07-29:
Piggy-backing this discussion, I noticed that the cached version of the database is more than 5 years old `[sam-library-cec-inverters-2019-03-05.csv](https://github.com/pvlib/pvlib-python/blob/83d90eb736d36028fb7028ae468703cfa45a3c5d/pvlib/data/sam-library-cec-inverters-2019-03-05.csv)` and `[sam-library-cec-modules-2019-03-05.csv](https://github.com/pvlib/pvlib-python/blob/83d90eb736d36028fb7028ae468703cfa45a3c5d/pvlib/data/sam-library-cec-modules-2019-03-05.csv)`. Is this something that could be updated on the newest SAM release [here](https://github.com/NREL/SAM/tree/develop/deploy/libraries), which is less than 1 year old? I could assist with that.

cwhanse, 2024-07-29:
> Piggy-backing this discussion, I noticed that the cached version of the CEC tables is more than 5 years old `data/sam-library-cec-inverters-2019-03-05.csv` and `data/sam-library-cec-modules-2019-03-05.csv`. Is this something that could be updated on the newest SAM release [here](https://github.com/NREL/SAM/tree/develop/deploy/libraries), which is less than 1 year old? I could assist with that.

I suppose we should update those cached files. 

I don't see a re-engineering of the CEC-SAM-pvlib parameter process on the horizon, unfortunately. #1345 #769 are relevant.