pvlib/pvlib-python #2154: Minor documentation error in parameter description with `pvlib.pvsystem.calcparams_cec` and `pvlib.pvsystem.calcparams_desoto`

kandersolar, 2024-08-06:
The `dEgdT` parameter in these two functions is currently documented as having units of `1/K`:

`calcparams_desoto`:

https://github.com/pvlib/pvlib-python/blob/72185cbc3be89e280b89a71da1c684b8fe5c15fd/pvlib/pvsystem.py#L1531-L1533

And `calcparams_cec`:

https://github.com/pvlib/pvlib-python/blob/72185cbc3be89e280b89a71da1c684b8fe5c15fd/pvlib/pvsystem.py#L1747-L1749

The correct units are `eV/K`, not `1/K`.  To fix this issue, these two documentation sentences should be updated to say `eV/K`.

I suggest we hold off on fixing this issue so that it remains available for a new contributor to fix during the upcoming pvlib user's group meeting in Copenhagen.

cwhanse, 2024-08-06:
I disagree.

` E_g = EgRef * (1 + dEgdT*(Tcell_K - Tref_K))` from Eq. 5 of Dobos (2012).

kandersolar, 2024-08-06:
Oops.  I got mixed up reading another reference.  Darn!  I was happy to have another easy PR opportunity for the user group meeting...

Closing as invalid.

closed 2024-08-06 by kandersolar