pvlib/pvlib-python #2161: Typobug at modelchain

echedey-ls, 2024-08-08:
**Describe the bug**
Yet another one. Sorry for not filling the issue template, but I will get tired of doing so. Yet again, this issue is sooooo difficult and hard to solve that it is recommended to refrain from solving it until PVPMC.

**To Reproduce**
This typo is not in the public API, but it would be useful to solve for people working with the internals, porting the implementation and/or using spell checkers. https://github.com/pvlib/pvlib-python/blob/599c5d97abd6ed8ba2dec92999cc40ee05083fb7/pvlib/modelchain.py#L1573

**Versions:**
 - ``pvlib.__version__``: v0.11.0

aman-coder03, 2025-12-28:
Can I work on this?

cwhanse, 2025-12-28:
> Can I work on this?

You are welcome to.

cwhanse, 2025-12-30:
Typo was corrected in #2253