pvlib/pvlib-python #2163: Typo at pvlib.pvsystem.calcparams_desoto

echedey-ls, 2024-08-08:
**Describe the bug**
This issue is very difficult, so GitHub please refrain from recommending it. Maybe it can be fix at PVPMC.

**To Reproduce**
Docs, see "saturation_current" parameter, https://pvlib-python.readthedocs.io/en/stable/reference/generated/pvlib.pvsystem.calcparams_desoto.html
File https://github.com/pvlib/pvlib-python/blob/599c5d97abd6ed8ba2dec92999cc40ee05083fb7/pvlib/pvsystem.py#L1553

**Expected behavior**
To read `current` instead of `curent` in the description.

**Versions:**
 - ``pvlib.__version__``: 0.11.0

cwhanse, 2024-08-08:
> This issue is very difficult, 

?

echedey-ls, 2024-08-08:
My bad for the lack of context, I thought many more of you were aware. I'm finding small issues that can be the starting point for a small how-to-contribute to pvlib workshop at EU PVPMC. Added warnings to all of these issues thx to @AdamRJensen .

adriesse, 2024-08-09:
> My bad for the lack of context, I thought many more of you were aware.

Not your bad. The core team has not been informed about overall plans for GSoC or workshop.

EduardoSarquis, 2024-08-22:
I'm working on it!

PR #2174 (2024-08-22): fix typo on calcparams_desoto help
There's a typo in the documentation of the "saturation_current" parameter on line 1553.

 - [x] Closes #2163
 - [x] I am familiar with the [contributing guidelines](https://pvlib-python.readthedocs.io/en/latest/contributing.html)
 - [x] Adds description and name entries in the appropriate "what's new" file in [`docs/sphinx/source/whatsnew`](https://github.com/pvlib/pvlib-python/tree/main/docs/sphinx/source/whatsnew) for all changes. Includes link to the GitHub Issue with `` :issue:`num` `` or this Pull Request with `` :pull:`num` ``. Includes contributor name [... cut]
closed 2024-08-22 by kandersolar