pvlib/pvlib-python #2172: Simple closed-form solution for alpha in detect_clearsky

agodbehere, 2024-08-21:
### Discussed in https://github.com/pvlib/pvlib-python/discussions/2171

<div type='discussions-op-text'>

<sup>Originally posted by **agodbehere** August 20, 2024</sup>
The algorithm in clearsky.detect_clearsky optimizes for a scaling parameter alpha by finding a value that minimizes the root mean-squared error (RMSE) between measurements and a scaled clear-sky model. Currently, the implementation computes the RMSE and uses scipy.optimize to iteratively minimize the function. However, a simple closed-form solution exists, bypassing the need to call scipy.optimize.

The objective function is $f(\alpha) = \sqrt{\frac{\sum_i (G_i - \alpha C_i)^2}{n}}$, where $G$ represents measurements of GHI and $C$ represents the clear-sky model.

Since $f(\alpha)$ is nonnegative, $\arg\min_\alpha f(\alpha) = \arg\min_\alpha f^2(\alpha)$. The value $n$ is a constant and similarly does not affect the solution. 

Thus, we end up with a modified objective function $\hat{f}(\alpha) = \sum_i (G_i-\alpha C_i)^2$. Solving for $\frac{d\hat{f}}{d\alpha} = 0$ yields the optimal value: $\hat{\alpha} = \frac{\sum_i G_iC_i}{\sum_i C_i^2}$. The result is nonnegative, so the constraint $\alpha > 0$ is satisfied implicitly. 
</div>

yhkee0404, 2024-09-28:
Duplicate of #2216

cwhanse, 2024-09-28:
Closed via #2217

closed 2024-09-28 by cwhanse