pvlib/pvlib-python #2183: Missing Descriptions for Formula Components in `haydavies` Function

BernatNicolau, 2024-08-26:
The formula components for the [`haydavies`](https://pvlib-python.readthedocs.io/en/stable/reference/generated/pvlib.irradiance.haydavies.html#pvlib.irradiance.haydavies) function seems to not be properly defined. The current formula is:

$I_{d} = DHI \left( A R_b + (1 - A) \left(\frac{1 + \cos \beta}{2}\right) \right)$

However, the documentation does not provide definitions or explanations for the components $A$ and $R_b$. While $DHI$ and $\beta$ are relatively straightforward, I feel it would be beneficial to include their descriptions for completeness. 

Additionally, [`isotropic`](https://pvlib-python.readthedocs.io/en/stable/reference/generated/pvlib.irradiance.isotropic.html), [`klutcher`](https://pvlib-python.readthedocs.io/en/stable/reference/generated/pvlib.irradiance.klucher.html) and [`reindl`](https://pvlib-python.readthedocs.io/en/stable/reference/generated/pvlib.irradiance.reindl.html)   also lack description of formula components.

#### Request

1. Update the documentation for the [`haydavies`](https://pvlib-python.readthedocs.io/en/stable/reference/generated/pvlib.irradiance.haydavies.html#pvlib.irradiance.haydavies) function to include descriptions for the following components:
   - $A$
   - $R_b$
   - $DHI$
   - $\beta$

2. Update the documentation for the [`isotropic`](https://pvlib-python.readthedocs.io/en/stable/reference/generated/pvlib.irradiance.isotropic.html), [`klutcher`](https://pvlib-python.readthedocs.io/en/stable/reference/generated/pvlib.irradiance.klucher.html) and [`reindl`](https://pvlib-python.readthedocs.io/en/stable/reference/generated/pvlib.irradiance.reindl.html) (shall different issues be created for each function?)

I think that providing these definitions will improve clarity.

Thank you!

AdamRJensen, 2024-08-26:
I do agree that it seems poorly documented and that the other variables should be defined. While we're at it, equations such as these belong in the Notes section of the documentation.

@BernatNicolau are you up for making a PR to improve the haydavies function?

BernatNicolau, 2024-08-26:
I will try to work on it but I don't think I will be able to prioritize it. I will post a comment when I start to work on the PR, so if anyone feels like doing it, go ahead :)

RDaxini, 2024-08-29:
I'd be happy to work on the documentation for at least one of these functions. I don't want to hijack your work here though @BernatNicolau , let me know if you were planning to make a start soon and/or whether my contribution would be helpful here.
It was nice to meet you at PVPMC btw, happy to see you are already very active now with pvlib! 

> shall different issues be created for each function?

I think this single issue is sufficient but separate PRs for each function might help with overall organisation of these developments.

RDaxini, 2024-08-30:
@BernatNicolau happy to hear the tutorial was helpful!

I have created a few PRs on this subject. Feel free to leave a review (another great type of contribution) if you have time! Your thoughts on the comments I have left or any of the code edits would be more than welcome. If you'd like some help with using any of the review features on github too then let us know.

PR #2191 (2024-09-24): Update irradiance.haydavies docs; use Wm⁻² instead of W/m^2 throughout `pvlib.irradiance`
 - [X] Partially closes #2183
 - [X] I am familiar with the [contributing guidelines](https://pvlib-python.readthedocs.io/en/latest/contributing.html)
 - [x] Adds description and name entries in the appropriate "what's new" file in [`docs/sphinx/source/whatsnew`](https://github.com/pvlib/pvlib-python/tree/main/docs/sphinx/source/whatsnew) for all changes. Includes link to the GitHub Issue with `` :issue:`num` `` or this Pull Request with `` :pull:`num` ``. Includes contributor name and/or GitHub username  [... cut]
PR #2192 (2024-09-26): Update irradiance.klucher docs
 - [X] Partially closes #2183, closes #1405
 - [X] I am familiar with the [contributing guidelines](https://pvlib-python.readthedocs.io/en/latest/contributing.html)
 - [x] Adds description and name entries in the appropriate "what's new" file in [`docs/sphinx/source/whatsnew`](https://github.com/pvlib/pvlib-python/tree/main/docs/sphinx/source/whatsnew) for all changes. Includes link to the GitHub Issue with `` :issue:`num` `` or this Pull Request with `` :pull:`num` ``. Includes contributor name and/or GitHub username (link with `` :ghuser:`user` ``).
 - [X] New  [... cut]
PR #2193 (2024-10-04): update irradiance.reindl docs
 - [X] Partially closes #2183, related to #1405
 - [X] I am familiar with the [contributing guidelines](https://pvlib-python.readthedocs.io/en/latest/contributing.html)
 - [x] Adds description and name entries in the appropriate "what's new" file in [`docs/sphinx/source/whatsnew`](https://github.com/pvlib/pvlib-python/tree/main/docs/sphinx/source/whatsnew) for all changes. Includes link to the GitHub Issue with `` :issue:`num` `` or this Pull Request with `` :pull:`num` ``. Includes contributor name and/or GitHub username (link with `` :ghuser:`user` ``).
 - [X] N [... cut]
closed 2024-09-24 by kandersolar