pvlib/pvlib-python #2185: Add method to fit PVsyst SDM to IEC 61853 data

kandersolar, 2024-08-27:
We have developed a method for determining parameter values for the PVsyst single diode model using IEC 61853-1 matrix measurements.  The method produces useful fits on a range of [IEC 61853 datasets](https://pvpmc.sandia.gov/datasets/pv-lifetime-module-datasets-clone/) without resorting to numerical optimization.  A journal article describing the method is nearing readiness for submission.  Once that manuscript is submitted, we plan to submit a PR to add the method to pvlib.  

We already have `pvlib.ivtools.sdm.fit_pvsyst_sandia`, which calculates SDM parameters based on I-V curves.  Perhaps new functions along these lines should encode the data source in their name, so this function could be called `fit_pvsyst_iec61853_sandia` or similar.

Note that such a method does not constitute a "PAN file generator" as it addresses only the electrical portion of a PAN file.

See also:
- #1935
- #2001

kandersolar, 2025-04-08:
The mentioned article is now available early and open access: https://doi.org/10.1109/JPHOTOV.2025.3554338

adriesse, 2025-04-11:
Would that have been substantially different or more difficult to do it for CEC and De Soto?

kandersolar, 2025-04-11:
Sure, I expect similar methods could be developed for CEC and De Soto.  It would probably be easier for those two than it was for the PVsyst SDM.

mikofski, 2025-04-12:
I’m pretty excited to start using this in pvfree to generate PVsyst parameter from SAM module libraries

cwhanse, 2025-04-12:
> I’m pretty excited to start using this in pvfree to generate PVsyst parameter from SAM module libraries

@mikofski I'm curious why the existing `pvsyst` functions wouldn't work for that? FYI, we expect #2212 to move ahead soon. It would be quite interesting to see a comparison among these somewhat independently derived techniques.

mikofski, 2025-04-12:
I was intending to use the existing functions, but I haven’t read the paper on these new regressions yet, so perhaps naively thought it might be a better option.

PR #2429 (2025-04-11): Add function for estimating PVsyst SDM parameters from IEC 61853-1 data
 - [x] Closes #2185
 - [x] I am familiar with the [contributing guidelines](https://pvlib-python.readthedocs.io/en/latest/contributing/index.html)
 - [x] Tests added
 - [x] Updates entries in [`docs/sphinx/source/reference`](https://github.com/pvlib/pvlib-python/blob/main/docs/sphinx/source/reference) for API changes.
 - [x] Adds description and name entries in the appropriate "what's new" file in [`docs/sphinx/source/whatsnew`](https://github.com/pvlib/pvlib-python/tree/main/docs/sphinx/source/whatsnew) for all changes. I [... cut]