pvlib/pvlib-python #2194: Example gallery suggestion: `spectrum.average_photon_energy`

RDaxini, 2024-09-01:
I am wondering what level of interest there is for an example gallery addition demonstrating the use of `spectrum.average_photon_energy`. 

I think an example calculation of the APE of several spectra could help users understand how to interpret the parameter (high values => blue-shifted spectrum, lower values => red-shifted spectrum) as well as guide users on how to use the function of course.

Since we have the `spectrum.spctrl2` function to simulate spectra, but the output is not usable directly as an input into `spectrum.average_photon_energy`, I think some example code linking the two would also be helpful so that users have a guide on how to simulate and characterise spectral distributions all within pvlib-python.

I'd be happy to create this example if the community thinks it'd be useful.

AdamRJensen, 2024-09-01:
That sounds great to me!

RDaxini, 2024-09-04:
I'll start working on this soon then and create a PR once I have something ready 👍

PR #2206 (2024-09-24): add `spectrum.average_photon_energy` gallery example
 - [X] Closes #2194
 - [X] I am familiar with the [contributing guidelines](https://pvlib-python.readthedocs.io/en/latest/contributing.html)
 - [ ] Tests added
 - [ ] Updates entries in [`docs/sphinx/source/reference`](https://github.com/pvlib/pvlib-python/blob/main/docs/sphinx/source/reference) for API changes.
 - [x] Adds description and name entries in the appropriate "what's new" file in [`docs/sphinx/source/whatsnew`](https://github.com/pvlib/pvlib-python/tree/main/docs/sphinx/source/whatsnew) for all changes. Includes link to the GitHu [... cut]
closed 2024-09-24 by kandersolar