pvlib/pvlib-python #2203: Dynamic albedo for floating PV

IoannisSifnaios, 2024-09-10:
Per @lmicheli recommendation, I suggest we add the dynamic albedo calculation as described [here](https://www.sciencedirect.com/science/article/pii/S2667113122000043).

I am up for taking the task of adding it to pvlib, but I would like to hear your thoughts on this addition.

kandersolar, 2024-09-10:
One concern around the white cap fraction defined by Eq 6: this paper seems to treat Eq 6 as returning a unitless fraction (it is subtracted from 1 in several places), but the cited reference claims that the equation returns values in percent.  That seems like a nontrivial error to me. 

More generally, would people actually use a spectral albedo model for FPV?  I think of spectral albedo as being pretty niche for ground PV (#1826), but maybe it's different in FPV?

cwhanse, 2024-09-10:
I think someone modeling floating PV would look for an albedo because the transposition models require one. The paper proposes a spectral albedo model because one of its sources provides a spectral refractive index. If we implement this method, I think we should add an integrated broadband albedo to the function's output.

I'm a bit skeptical of albedo models that are calibrated to satellite observations, as the components of this model appear to be. I can see how foam, whitecaps and surface roughness could affect the large-area albedo viewed from space. How would these features affect the albedo "seen" by a floating PV system? More generally, is an accurate albedo important in a performance model for ocean floating PV? I imagine the modules would be at quite shallow tilt on the floaters, with their view to the horizon mostly blocked by adjacent floaters. If that's right, the far-field sea-reflected irradiance would be a negligible factor in system power.

echedey-ls, 2024-09-10:
> I think we should add an integrated broadband albedo to the function's output.

I would prefer to leave that to another function, so it's code can be reused if spectral data is extended in the future and the implementation and docs of this model are more compact.

IoannisSifnaios, 2024-09-11:
> I scanned the paper but did not see any validation of the model.

Good point @adriesse! There is also [this paper](https://opg.optica.org/oe/fulltext.cfm?uri=oe-19-27-26429&id=225797) that has a fairly similar model (and also a validation section). However, based on the previous comments, it is still not entirely clear to me if such a model would be a valuable addition.