pvlib/pvlib-python #2210: Split the contributing page

RDaxini, 2024-09-13:
**Is your feature request related to a problem? Please describe.**
I think that [this single contributing page](https://pvlib-python.readthedocs.io/en/stable/contributing.html) in its current form may be overwhelming for new users; it certainly was for me not so many months ago 😅

**Describe the solution you'd like**
After reading some of the contributing pages on other projects (through the discussion in https://github.com/pvlib/pvlib-python/issues/2202), I feel like our contributing page would benefit from being broken up into several pages.

These pages would still all be under the same "Contributing" section, but they could be navigated through using the sidebar (like the user guide?)

In this way, we could keep the really basic basics on one (the first) page, then move other details such as code style, testing, virtual environments, etc. onto separate pages rather than just separate sections on the same page.

I think having the new pages split somewhere along the lines of the current sections would be a good starting point. I am interested to hear if anyone has any other suggestions for a structure, or for any entirely new sections.

**Additional context**
There are ongoing discussions on standardising the reference style (#2202) and standardising the typesetting of units (#2205). I think that the new (split) contributing section could include a style guide page with all of this new information on documentation style, as well as existing style information related to code style that we already have.

I am +1 for @AdamRJensen's suggestion to include a template function and I think the aforementioned style guide section would be a good place for this.

One potential new section (besides the style guide) could be a "Contact" page (later discussion in #2200) where readers are advised to create a new issue if they want to reach out. I am thinking a dedicated new page rather than a sentence in another page might be better because a couple of screenshots showing how to create a github issue might be helpful for new users.

If the community thinks this idea for restructuring would be useful, I'd be happy to have a crack at it.

RDaxini, 2024-09-13:
> **Is your feature request related to a problem? Please describe.** I think that [this single contributing page](https://pvlib-python.readthedocs.io/en/stable/contributing.html) in its current form may be overwhelming for new users; it certainly was for me not so many months ago 😅
> 
> **Describe the solution you'd like** After reading some of the contributing pages on other projects (through the discussion in #2202), I feel like our contributing page would benefit from being broken up into several pages.
> 
> These pages would still all be under the same "Contributing" section, but they could be navigated through using the sidebar (like the user guide?)
> 
> In this way, we could keep the really basic basics on one (the first) page, then move other details such as code style, testing, virtual environments, etc. onto separate pages rather than just separate sections on the same page.
> 
> I think having the new pages split somewhere along the lines of the current sections would be a good starting point. I am interested to hear if anyone has any other suggestions for a structure, or for any entirely new sections.
> 
> **Additional context** There are ongoing discussions on standardising the reference style (#2202) and standardising the typesetting of units (#2205). I think that the new (split) contributing section could include a style guide page with all of this new information on documentation style, as well as existing style information related to code style that we already have.
> 
> I am +1 for @AdamRJensen's suggestion to include a template function and I think the aforementioned style guide section would be a good place for this.
> 
> One potential new section (besides the style guide) could be a "Contact" page (later discussion in #2200) where readers are advised to create a new issue if they want to reach out. I am thinking a dedicated new page rather than a sentence in another page might be better because a couple of screenshots showing how to create a github issue could be included to help new users.
> 
> If the community thinks this idea for restructuring would be useful, I'd be happy to have a crack at it.

kandersolar, 2024-09-16:
+1 to splitting it up.  I would keep the "This documentation" and "Code of Conduct" sections on the first page instead of splitting them into their own pages.

echedey-ls, 2024-09-20:
Mind if I work on this? I want to propose more changes to the contribution guidelines, thou I will keep that in a separate PR unless you tell me otherwise.

RDaxini, 2024-09-20:
@echedey-ls ah I have made a start locally already, just not opened the PR(s) on GitHub yet.
To speed up the process (possibly make reviews easier too) I could keep the first PR focused on splitting up the page, then if you have new suggestions for the contributing guidelines you can start revising what's already there without a problem. I can move the creation of new section(s)/text, such as documentation style guide for references that I proposed, into a separate PR after the splitting.
How does that sound?

echedey-ls, 2024-09-20:
Sounds reasonable and wonderful. I suppose there are other views on doing so many changes in one PR, but I prefer to start with the brainstorm early (will be moving that to #2067 and #2081 once I gather all my ideas).

RDaxini, 2024-09-23:
> a "Contact" page (later discussion in #2200) where readers are advised to create a new issue if they want to reach out. 

Would this be best placed as:
a) a new page within the contributing section
b) a new page within some other section
c) a new section (new tab at the top of https://pvlib-python.readthedocs.io/en/stable/)
d) a line or two on some other page within the contributing (or other) section
e) not existing

What I think this addition would include is couple of lines two encouraging people to get in touch if they have any questions, and links to the Github repo (or direct to issues page) and the Google page.  Besides the obvious benefit of inviting engagement, it could also be a way to nudge non-Github users (potential contributors) onto Github 👀

RDaxini, 2024-09-27:
> > a "Contact" page (later discussion in #2200) where readers are advised to create a new issue if they want to reach out.
> 
> Would this be best placed as: a) a new sub-page within the contributing page group b) a new sub-page within some other page group c) a new page group (new tab at the top of https://pvlib-python.readthedocs.io/en/stable/) d) a line or two on some other sub-page within the contributing (or other) page group e) not existing
> 
> What I think this addition would include is couple of lines two encouraging people to get in touch if they have any questions, and links to the Github repo (or direct to issues page) and the Google page. Besides the obvious benefit of inviting engagement, it could also be a way to nudge non-Github users (potential contributors) onto Github 👀
> 
> A new tab at the top appeals to me but it seems like a big deal for very little (although important) content so I am 60/40 leaning in that direction. I think my next preference would be a subpage within the (new) contributing page group. I am interested to hear what others think.

This may be resolved by #2224, so perhaps I should close this issue as completed by #2219?

closed 2024-09-27 by RDaxini