pvlib/pvlib-python #2221: PVGIS tests are failing

kandersolar, 2024-09-23:
Several PVGIS tests are failing because the expected values no longer match what is returned by the API.  See an example CI run: https://github.com/pvlib/pvlib-python/actions/runs/11003008118/job/30551180154#step:6:256

```
=========================== short test summary info ============================
FAILED pvlib/tests/iotools/test_pvgis.py::test_get_pvgis_tmy - assert False
 +  where False = <function allclose at 0x7fbb5aafbf70>(time(UTC)\n2014-01-01 00:00:00+00:00    1.08\n2014-01-01 01:00:00+00:00    0.42\n2014-01-01 02:00:00+00:00   -0.24\n2014-0...:00    3.05\n2018-12-31 22:00:00+00:00    2.39\n2018-12-31 23:00:00+00:00    1.73\nName: T2m, Length: 8760, dtype: float64, time(UTC)\n2009-01-01 00:00:00+00:00    2.29\n2009-01-01 01:00:00+00:00    2.10\n2009-01-01 02:00:00+00:00    1.91\n2009-0...:00    2.87\n2011-12-31 22:00:00+00:00    2.67\n2011-12-31 23:00:00+00:00    2.48\nName: T2m, Length: 8760, dtype: float64)
 +    where <function allclose at 0x7fbb5aafbf70> = np.allclose
FAILED pvlib/tests/iotools/test_pvgis.py::test_get_pvgis_tmy_kwargs - assert False
 +  where False = <function allclose at 0x7fbb5aafbf70>(time(UTC)\n2014-01-01 00:00:00+00:00    0.0\n2014-01-01 01:00:00+00:00    0.0\n2014-01-01 02:00:00+00:00    0.0\n2014-01-0...00:00    0.0\n2018-12-31 22:00:00+00:00    0.0\n2018-12-31 23:00:00+00:00    0.0\nName: G(h), Length: 8760, dtype: float64, array([0., 0., 0., ..., 0., 0., 0.]))
 +    where <function allclose at 0x7fbb5aafbf70> = np.allclose
 +    and   array([0., 0., 0., ..., 0., 0., 0.]) = 0       0.0\n1       0.0\n2       0.0\n3       0.0\n4       0.0\n       ... \n8755    0.0\n8756    0.0\n8757    0.0\n8758    0.0\n8759    0.0\nName: G(h), Length: 8760, dtype: float64.values
FAILED pvlib/tests/iotools/test_pvgis.py::test_get_pvgis_tmy_basic - assert False
 +  where False = <function allclose at 0x7fbb5aafbf70>(time(UTC)\n2014-01-01 00:00:00+00:00    1.08\n2014-01-01 01:00:00+00:00    0.42\n2014-01-01 02:00:00+00:00   -0.24\n2014-0...:00    3.05\n2018-12-31 22:00:00+00:00    2.39\n2018-12-31 23:00:00+00:00    1.73\nName: T2m, Length: 8760, dtype: float64, time(UTC)\n2009-01-01 00:00:00+00:00    2.29\n2009-01-01 01:00:00+00:00    2.10\n2009-01-01 02:00:00+00:00    1.91\n2009-0...:00    2.87\n2011-12-31 22:00:00+00:00    2.67\n2011-12-31 23:00:00+00:00    2.48\nName: T2m, Length: 8760, dtype: float64)
 +    where <function allclose at 0x7fbb5aafbf70> = np.allclose
FAILED pvlib/tests/iotools/test_pvgis.py::test_get_pvgis_tmy_csv - assert False
 +  where False = <function allclose at 0x7fbb5aafbf70>(time(UTC)\n2014-01-01 00:00:00+00:00    1.08\n2014-01-01 01:00:00+00:00    0.42\n2014-01-01 02:00:00+00:00   -0.24\n2014-0...:00    3.05\n2018-12-31 22:00:00+00:00    2.39\n2018-12-31 23:00:00+00:00    1.73\nName: T2m, Length: 8760, dtype: float64, time(UTC)\n2009-01-01 00:00:00+00:00    2.29\n2009-01-01 01:00:00+00:00    2.10\n2009-01-01 02:00:00+00:00    1.91\n2009-0...:00    2.87\n2011-12-31 22:00:00+00:00    2.67\n2011-12-31 23:00:00+00:00    2.48\nName: T2m, Length: 8760, dtype: float64)
 +    where <function allclose at 0x7fbb5aafbf70> = np.allclose
FAILED pvlib/tests/iotools/test_pvgis.py::test_get_pvgis_tmy_epw - assert False
 +  where False = <function allclose at 0x7fbb5aafbf70>(2014-01-01 00:00:00+01:00    0.0\n2014-01-01 01:00:00+01:00    0.0\n2014-01-01 02:00:00+01:00    0.0\n2014-01-01 03:00:00...+01:00    0.0\n2018-12-31 22:00:00+01:00    0.0\n2018-12-31 23:00:00+01:00    0.0\nName: ghi, Length: 8760, dtype: float64, time(UTC)\n2009-01-01 00:00:00+00:00    0.0\n2009-01-01 01:00:00+00:00    0.0\n2009-01-01 02:00:00+00:00    0.0\n2009-01-0...00:00    0.0\n2011-12-31 22:00:00+00:00    0.0\n2011-12-31 23:00:00+00:00    0.0\nName: G(h), Length: 8760, dtype: float64)
 +    where <function allclose at 0x7fbb5aafbf70> = np.allclose
 +    and   2014-01-01 00:00:00+01:00    0.0\n2014-01-01 01:00:00+01:00    0.0\n2014-01-01 02:00:00+01:00    0.0\n2014-01-01 03:00:00...+01:00    0.0\n2018-12-31 22:00:00+01:00    0.0\n2018-12-31 23:00:00+01:00    0.0\nName: ghi, Length: 8760, dtype: float64 =                            year  ...  liquid_precipitation_quantity\n2014-01-01 00:00:00+01:00  2014  ...              ...                       99\n2018-12-31 23:00:00+01:00  2018  ...                             99\n\n[8760 rows x 35 columns].ghi
======= 5 failed, 138 passed, 8 warnings, 25 rerun in 253.54s (0:04:13) ========
```

I suppose it has something to do with the release of PVGIS 5.3.  See here:
https://joint-research-centre.ec.europa.eu/photovoltaic-geographical-information-system-pvgis/pvgis-releases/pvgis-53_en

AdamRJensen, 2024-09-23:
I think this is good news!

I'll prepare some updated tests soon

kandersolar, 2024-10-15:
@AdamRJensen PVGIS tests are failing again: https://github.com/pvlib/pvlib-python/actions/runs/11354441803/job/31581700027

```
=========================== short test summary info ============================
FAILED pvlib/tests/iotools/test_pvgis.py::test_get_pvgis_tmy - assert False
 +  where False = <function allclose at 0x7fd219cd2cf0>(time(UTC)\n2018-01-01 00:00:00+00:00    2.04\n2018-01-01 01:00:00+00:00    1.98\n2018-01-01 02:00:00+00:00    1.92\n2018-0...:00    2.23\n2016-12-31 22:00:00+00:00    2.17\n2016-12-31 23:00:00+00:00    2.10\nName: T2m, Length: 8760, dtype: float64, time(UTC)\n2014-01-01 00:00:00+00:00    1.08\n2014-01-01 01:00:00+00:00    0.42\n2014-01-01 02:00:00+00:00   -0.24\n2014-0...:00    3.05\n2018-12-31 22:00:00+00:00    2.39\n2018-12-31 23:00:00+00:00    1.73\nName: T2m, Length: 8760, dtype: float64)
 +    where <function allclose at 0x7fd219cd2cf0> = np.allclose
FAILED pvlib/tests/iotools/test_pvgis.py::test_get_pvgis_tmy_kwargs - assert False
 +  where False = <function allclose at 0x7fd219cd2cf0>(time(UTC)\n2018-01-01 00:00:00+00:00    0.0\n2018-01-01 01:00:00+00:00    0.0\n2018-01-01 02:00:00+00:00    0.0\n2018-01-0...00:00    0.0\n2016-12-31 22:00:00+00:00    0.0\n2016-12-31 23:00:00+00:00    0.0\nName: G(h), Length: 8760, dtype: float64, array([0, 0, 0, ..., 0, 0, 0]))
 +    where <function allclose at 0x7fd219cd2cf0> = np.allclose
 +    and   array([0, 0, 0, ..., 0, 0, 0]) = 0       0\n1       0\n2       0\n3       0\n4       0\n       ..\n8755    0\n8756    0\n8757    0\n8758    0\n8759    0\nName: G(h), Length: 8760, dtype: int64.values
FAILED pvlib/tests/iotools/test_pvgis.py::test_get_pvgis_tmy_basic - assert False
 +  where False = <function allclose at 0x7fd219cd2cf0>(time(UTC)\n2018-01-01 00:00:00+00:00    2.04\n2018-01-01 01:00:00+00:00    1.98\n2018-01-01 02:00:00+00:00    1.92\n2018-0...:00    2.23\n2016-12-31 22:00:00+00:00    2.17\n2016-12-31 23:00:00+00:00    2.10\nName: T2m, Length: 8760, dtype: float64, time(UTC)\n2014-01-01 00:00:00+00:00    1.08\n2014-01-01 01:00:00+00:00    0.42\n2014-01-01 02:00:00+00:00   -0.24\n2014-0...:00    3.05\n2018-12-31 22:00:00+00:00    2.39\n2018-12-31 23:00:00+00:00    1.73\nName: T2m, Length: 8760, dtype: float64)
 +    where <function allclose at 0x7fd219cd2cf0> = np.allclose
FAILED pvlib/tests/iotools/test_pvgis.py::test_get_pvgis_tmy_csv - assert False
 +  where False = <function allclose at 0x7fd219cd2cf0>(time(UTC)\n2018-01-01 00:00:00+00:00    2.04\n2018-01-01 01:00:00+00:00    1.98\n2018-01-01 02:00:00+00:00    1.92\n2018-0...:00    2.23\n2016-12-31 22:00:00+00:00    2.17\n2016-12-31 23:00:00+00:00    2.10\nName: T2m, Length: 8760, dtype: float64, time(UTC)\n2014-01-01 00:00:00+00:00    1.08\n2014-01-01 01:00:00+00:00    0.42\n2014-01-01 02:00:00+00:00   -0.24\n2014-0...:00    3.05\n2018-12-31 22:00:00+00:00    2.39\n2018-12-31 23:00:00+00:00    1.73\nName: T2m, Length: 8760, dtype: float64)
 +    where <function allclose at 0x7fd219cd2cf0> = np.allclose
FAILED pvlib/tests/iotools/test_pvgis.py::test_get_pvgis_tmy_epw - assert False
 +  where False = <function allclose at 0x7fd219cd2cf0>(2018-01-01 00:00:00+01:00    0.0\n2018-01-01 01:00:00+01:00    0.0\n2018-01-01 02:00:00+01:00    0.0\n2018-01-01 03:00:00...+01:00    0.0\n2016-12-31 22:00:00+01:00    0.0\n2016-12-31 23:00:00+01:00    0.0\nName: ghi, Length: 8760, dtype: float64, time(UTC)\n2014-01-01 00:00:00+00:00    0.0\n2014-01-01 01:00:00+00:00    0.0\n2014-01-01 02:00:00+00:00    0.0\n2014-01-0...00:00    0.0\n2018-12-31 22:00:00+00:00    0.0\n2018-12-31 23:00:00+00:00    0.0\nName: G(h), Length: 8760, dtype: float64)
 +    where <function allclose at 0x7fd219cd2cf0> = np.allclose
 +    and   2018-01-01 00:00:00+01:00    0.0\n2018-01-01 01:00:00+01:00    0.0\n2018-01-01 02:00:00+01:00    0.0\n2018-01-01 03:00:00...+01:00    0.0\n2016-12-31 22:00:00+01:00    0.0\n2016-12-31 23:00:00+01:00    0.0\nName: ghi, Length: 8760, dtype: float64 =                            year  ...  liquid_precipitation_quantity\n2018-01-01 00:00:00+01:00  2018  ...              ...                       99\n2016-12-31 23:00:00+01:00  2016  ...                             99\n\n[8760 rows x 35 columns].ghi
======= 5 failed, 138 passed, 8 warnings, 25 rerun in 238.69s (0:03:58) ========
```

PR #2222 (2024-09-26): Update PVGIS TMY tests and files

 - [x] Closes #2221
 - [x] I am familiar with the [contributing guidelines](https://pvlib-python.readthedocs.io/en/latest/contributing.html)
 - [x] Tests added
 - [x] Adds description and name entries in the appropriate "what's new" file in [`docs/sphinx/source/whatsnew`](https://github.com/pvlib/pvlib-python/tree/main/docs/sphinx/source/whatsnew) for all changes. Includes link to the GitHub Issue with `` :issue:`num` `` or this Pull Request with `` :pull:`num` ``. Includes contributor name and/or GitHub username (link with `` :ghuser:`user` ``).
 - [x] New co [... cut]
PR #2305 (2024-11-27): Fix PVGIS TMY tests
 - [x] Closes #2221
 - [x] I am familiar with the [contributing guidelines](https://pvlib-python.readthedocs.io/en/latest/contributing.html)
 - [x] Tests added
 - ~[ ] Updates entries in [`docs/sphinx/source/reference`](https://github.com/pvlib/pvlib-python/blob/main/docs/sphinx/source/reference) for API changes.~
 - [x] Adds description and name entries in the appropriate "what's new" file in [`docs/sphinx/source/whatsnew`](https://github.com/pvlib/pvlib-python/tree/main/docs/sphinx/source/whatsnew) for all changes. Includes link to the GitHub Issue with `` :issue:`num` `` [... cut]
closed 2024-11-27 by kandersolar