pvlib/pvlib-python #2231: Input parameter naming convention - time vs times

AdamRJensen, 2024-09-28:
Several pvlib functions take an index as one of the input parameters, however, it is inconsistent whether this parameter is called `times` or `time`.

## See a few examples below of `TIME`:

https://github.com/pvlib/pvlib-python/blob/4cfda4a14366217ba9bb3b15d2531c61b7507e69/pvlib/solarposition.py#L30
https://github.com/pvlib/pvlib-python/blob/4cfda4a14366217ba9bb3b15d2531c61b7507e69/pvlib/solarposition.py#L128
https://github.com/pvlib/pvlib-python/blob/4cfda4a14366217ba9bb3b15d2531c61b7507e69/pvlib/tools.py#L120

## Here's a few examples of `TIMES`:
https://github.com/pvlib/pvlib-python/blob/4cfda4a14366217ba9bb3b15d2531c61b7507e69/pvlib/tools.py#L424
https://github.com/pvlib/pvlib-python/blob/4cfda4a14366217ba9bb3b15d2531c61b7507e69/pvlib/solarposition.py#L392
https://github.com/pvlib/pvlib-python/blob/4cfda4a14366217ba9bb3b15d2531c61b7507e69/pvlib/solarposition.py#L1343

RDaxini, 2024-09-28:
Related: #1253
In particular [this comment](https://github.com/pvlib/pvlib-python/issues/1253#issuecomment-2344790740) (btw I have updated the table with time/times now thanks to this issue)

adriesse, 2024-10-01:
I vote for `time`.

AdamRJensen, 2024-10-01:
I personally prefer times 😋

kandersolar, 2024-10-01:
A good guiding principle may be to use plural words only when multiple values are necessary.  For example, I think `times` is an appropriate name for the `_get_sample_intervals(times, win_length)` function, since it actually requires multiple timestamps.   For other functions (e.g. `hour_angle`) that work fine with single timestamps, would we say that the function returns `hour_angles`, and do the rise/set functions return `sunrises` and `sunsets`?  Here I think singular makes sense.

An argument against `time` in general is that it has the same name as a python standard library.

cwhanse, 2024-10-01:
> An argument against time in general is that it has the same name as a python standard library.

Good point. I overlooked this when agreeing with "time" with the singular/plural grammar in mind. I retract my vote; pvlib should not shadow any standard python names.

adriesse, 2024-10-02:
> > An argument against time in general is that it has the same name as a python standard library.
> 
> Good point. I overlooked this when agreeing with "time" with the singular/plural grammar in mind. I retract my vote; pvlib should not shadow any standard python names.

As the shadowing only occurs within the function, this doesn't seem so serious.  It will not cause any problems for the user of the function.

cwhanse, 2024-10-02:
> As the shadowing only occurs within the function, this doesn't seem so serious. It will not cause any problems for the user of the function.

Yes, but we can't anticipate how a user structures their code. Maybe

echedey-ls, 2024-10-02:
> > As the shadowing only occurs within the function, this doesn't seem so serious. It will not cause any problems for the user of the function.
> 
> Yes, but we can't anticipate how a user structures their code.

I don't see an issue and fully agree with @adriesse on that it is not a problem. Can you provide an example of what you have in mind? AFAIK, this parameter will only override `time` identifier inside its namespace, the pvlib func.

cwhanse, 2024-10-02:
I can't provide an example where code fails to run. And we've had `time` as a function parameter for years, so I yield my objection.

echedey-ls, 2024-10-02:
No problem.

echedey-ls, 2024-10-02:
I was about to edit this function:
https://github.com/pvlib/pvlib-python/blob/4cfda4a14366217ba9bb3b15d2531c61b7507e69/pvlib/solarposition.py#L392

in #2237 to add another example and test that all works flawlessly, but reading through its documentation I think the appropriate rename is `date` instead of `times`. From the docs (and code) of the internal function it uses:

https://github.com/pvlib/pvlib-python/blob/29b1f86e515bdf6b0b2bf9fc0addafb10646f572/pvlib/spa.py#L1106

These "`times`" must be at 00:00. Funny, isn't it?

Feel free to weight in if there are other words to be considered that I've missed.

adriesse, 2024-10-03:
No opinion here for pvlib, but in my own work I standardized on `time` for many dates ago.  I see `date` simply as low-resolution `time`, with the time-of-day unspecified.

echedey-ls, 2024-10-03:
> I see `date` simply as low-resolution `time`, with the time-of-day unspecified.

That did convince me.

echedey-ls, 2024-10-11:
Shall `transit_sunrise_sunset`:

https://github.com/pvlib/pvlib-python/blob/cbe4cc5a8bb3bb529f2dd885aebcd527fedbbc63/pvlib/spa.py#L1106

be changed to `time` too? It requires the time part to be exactly at `00:00`.