pvlib/pvlib-python #2233: DNI/DHI estimation

eshnEnergiDanmark, 2024-09-29:
I have weather data of GNI, but I need to estimate the DNI and DHI which I have done with the Erbs Drisse model:

from pvlib.irradiance import erbs_driesse, get_total_irradiance

dni_dhi = erbs_driesse(
    ghi=vejr["ghi"].values,
    zenith=sun_pos["zenith"].values,
    datetime_or_doy=pd.DatetimeIndex(sun_pos.index),
)

However I get some wierd output when plotting the data in a very clear sky period. How come? It looks shifted and makes the simulation bad
[image]

markcampanelli, 2024-09-29:
A quick sanity check would be to run this same data through, say, the
original Erbs model and see if you get a similar artifact. That would
likely catch an error in the solar position calculation or indexing.

Mark Campanelli
LinkedIn <https://www.linkedin.com/in/markcampanelli/>
Intelligent Measurement Systems LLC
<https://intelligentmeasurementsystems.com>
Try PVfit <https://pvfit.app> today!

eshnEnergiDanmark, 2024-09-29:
I get the same artifact with original Erbs. I also suspect it is an eropr in the solar position calculation, the way the plot looks. I have the following code

from pvlib.location import Location
vejr = pd.read_csv(...)
location = Location(lat, lon, tz="UTC")
sun_pos = location.get_solarposition(vejr.index)

AdamRJensen, 2024-09-29:
Your weather data is almost certainly in a different timezone than utc. ETC/UTC-1 should be correct

eshnEnergiDanmark, 2024-09-29:
I'm almost certain that my weather data is UTC. I have tried to play around with the sun_position and change the timezone of this to see how it impacts the outcome of Erbs-Driesse. I use this code: 

sun_pos.index = sun_pos.index.tz_convert("Etc/GMT-8")
sun_pos.index = sun_pos.index.tz_localize(None)

# %%
from pvlib.irradiance import erbs_driesse

dni_dhi = erbs_driesse(
    ghi=vejr["ghi"].values,
    zenith=sun_pos["zenith"].values,
    datetime_or_doy=pd.DatetimeIndex(sun_pos.index),
)

And get this plot, where I still have this wierd sharp rise in the start of every DNI-curve

[image]

eshnEnergiDanmark, 2024-09-29:
I am very certain my weather data is in UTC.

AdamRJensen, 2024-09-29:
> I am very certain my weather data is in UTC. 

I'd be happy to take a lot if you're up for sharing a piece of the data: arajen@dtu.dk

cwhanse, 2024-09-29:
> Your weather data is almost certainly in a different time zone.

Or the sun position times are not aligned with the weather data times.

eshnEnergiDanmark, 2024-09-29:
Update. I found that if I shifted +30 min to the weather data, I get nice plot of the estimated DNI/DHI. Here I have the ghi, dni, dhi for both the weather data (y) and clear sky model (x) - and it looks alot more nice
[image]

closed 2024-09-30 by eshnEnergiDanmark