pvlib/pvlib-python #2241: modeling a microinverter array

fradog, 2024-10-07:
I'm new to pvlib, and have successfully run through a few tutorials. My array uses microinverters and has three different orientations of arrays. I believe I've discovered that pvlib will not model this topology.  This is pretty tedious to represent with three separate runs and multiplying each by the number of modules, although certainly doable.

I find this to be an obvious feature that should be available given the popularity of microinverter arrays.

I had thought that setting up several arrays, where each had strings=the number of modules with a given orientation, and then in PVsystem setting modules_per_string=1 and strings_per_inverter=1   should have done the trick. The result however clips the output based on a single inverter.

I'm wondering why there's no inverters_per_module option?
Is this a difficult feature to implement?

Thanks!

cwhanse, 2024-10-07:
> I'm wondering why there's no inverters_per_module option?

There's no inverters_per_array option at present. A PVSystem can contain several Arrays but only one inverter. The design goes back to the start of pvlib. Supporting multiple inverters in a PVSystem is certainly possible, but non-trivial to implement, and could require a new concept for a power block, or similar, defined as the scope of the DC input to a single inverter. Then a PVSystem would have multiple power blocks, and each power block would have multiple Arrays.

Or something like that.

fradog, 2024-10-07:
Thanks for the explanation. I get it.
 I wonder if a workaround would be to multiply the inverter's max current by the number of modules?

cwhanse, 2024-10-07:
> multiply the inverter's max current by the number of modules?

The inverter models in pvlib don't have any current parameters.

I would define three arrays, each with one module and one inverter, then multiple the array AC power by the number of inverters.

fradog, 2024-10-08:
Thanks for the help. I went in and multiplied the power numbers for the inverter and got rid of the clipping. I'm not sure if that skews the results. I will check vs just multiplying the results. 
I really appreciate what you guys are doing. I'm learning. I will move my naive questions to a more appropriate forum.

fradog, 2024-10-08:
update, The power result I get by multiplying the power params from the cecinverter (enphase iq6) by the number of modules is exactly the same as if I multiply the ac results. So it seems to work pretty well. I'm including a comp of my home  pv system's output on 1/21/21 with the pvlib result and the code I thrashed based on the example for multiple arrays at different orientations. 
The only remaining issue: I can't get it to plot the array dc results all together in the same figure. It has something to do with the fact I replaced the module parameters from the example with the cecmodule parameters for my particular Q Cell, I think.

[image]

``
```
from pvlib import pvsystem, modelchain, location
import pandas as pd
import matplotlib.pyplot as plt

cecmodules = pvsystem.retrieve_sam('CECMod')
cecinverters = pvsystem.retrieve_sam('CECInverter')

module = cecmodules['Hanwha_Q_Cells_Q_PEAK_G4_1_290']
inverter = cecinverters['Enphase_Energy_Inc___IQ6_60_x_US__240V_']

nummods = 32.0

inverter.Paco=inverter.Paco*nummods
inverter.Pdco=inverter.Pdco*nummods
inverter.Idcmax=inverter.Idcmax*nummods

array_kwargs = dict(
    module_parameters=module,
    temperature_model_parameters=dict(a=-3.56, b=-0.075, deltaT=3)
)

arrays = [
    pvsystem.Array(pvsystem.FixedMount(20, 198), strings=23, name='S-Facing Array',
                   **array_kwargs),
    pvsystem.Array(pvsystem.FixedMount(20, 108), strings=2, name='E-Facing Array',
                   **array_kwargs),
    pvsystem.Array(pvsystem.FixedMount(20, 288), strings=7, name='W-Facing Array',
                  **array_kwargs),
]

loc = location.Location(34.17, -118.97)
system = pvsystem.PVSystem(arrays=arrays, inverter_parameters=inverter)
mc = modelchain.ModelChain(system, loc, aoi_model='physical',
                           spectral_model='no_loss')

times = pd.date_range('2019-01-01 6:00', '2019-01-01 20:00', freq='10min',
                      tz='America/los_Angeles')
weather = loc.get_clearsky(times)
mc.run_model(weather)
weather.plot(figsize=(16,9))
plt.show()

fig, ax = plt.subplots(sharex='all',sharey='all')
for array, pdc in zip(system.arrays, mc.results.dc):
    pdc.plot(label=f'{array.name}')
mc.results.ac.plot(label='Inverter')
plt.ylabel('System Output')
plt.legend()
plt.show()
maxACp = max(mc.results.ac)
print(maxACp)

closed 2024-10-08 by fradog

cwhanse, 2024-10-08:
Commenting for posterity:

1. the DC and AC results may be hard to see because DC results are for a single module and are at different scales than the AC power. Max AC power 6000W, max DC voltage is about 50V.
2. Technically this calculation is simulating a single 267W module feeding a 7.63kW inverter. Because only the inverter AC and DC power parameters are scaled, the inverter still turns on at the microinverter power level (about 1W). The CEC (Sandia) model is not exactly linear in the power parameters, so the scaling used here deviates from the fitted efficiency curves. Still, it may be close enough for the user's intent.

fradog, 2024-10-10:
On further inspection, I decided to output DC energy and AC energy and found that AC was higher than DC, which is of course an erroneous result. I presume it is due to the abovementioned inverter curve and scale issue. I wrote a new script that essentially multiplies the modelchain results of various arrays, each with one module and inverter, by the number of modules in each array. I now believe this result makes sense and still compares well with results for an actual clear day from my system.
Interestingly, it shows that my inverters are clipping on summer days when it's still cool enough but sunny enough to get nearer to the rated output of the panel. I had never noticed this on my Enphase reports but now see the curves are in fact clipped. Not a ton of loss so nothing to do about it, but I never would have noticed if not for pvlib. Thanks!