pvlib/pvlib-python #2255: `transit_sunrise_sunset`: seems as a public function, not in public docs?

echedey-ls, 2024-10-11:
**Describe the bug**
https://github.com/pvlib/pvlib-python/blob/cbe4cc5a8bb3bb529f2dd885aebcd527fedbbc63/pvlib/spa.py#L1106

Personally, I don't find any signs that it may be a private function. However, it does not appear in the webpage:

- https://pvlib-python.readthedocs.io/en/stable/search.html?q=transit_sunrise_sunset

Well it's not true: it appears here https://pvlib-python.readthedocs.io/en/stable/reference/generated/pvlib.spa.html
I wasn't around when `pvlib.spa` didn't get documented, so seeing it's not the only one, maybe it was deliberate? 

**Versions:**
 - ``pvlib.__version__``: 0.11.0

cwhanse, 2024-10-11:
In v0.1 when this function was added the documents didn't have the auto-built the API reference pages. Omission was probably to avoid cluttering the solarposition.py module page and only list the most significant functions.

cwhanse, 2024-10-21:
I don't see any downside to adding this function to the API documentation.

AdamRJensen, 2024-10-22:
Agreed, a PR to include it is welcomed 😄

closed 2026-10-28 by kandersolar