pvlib/pvlib-python #2272: Standardise variable name for clearsky GHI and add to glossary

RDaxini, 2024-10-20:
**Is your feature request related to a problem? Please describe.**
I have counted three variations of the term in the docs:

`clearsky` (in `clearsky.detect_clearsky`)
`clearsky_ghi` (in `irradiance.clearsky_index`)
`ghi_clearsky` (in `irradiance.dirindex`)
and no entry in the [variables and symbols page](https://pvlib-python.readthedocs.io/en/stable/user_guide/variables_style_rules.html) (soon to become a glossary #2234) 

**Describe the solution you'd like**
* Agree a single variable name
* Add an entry to the variables variables and symbols page

Consistency with # would be good.

I think `clearsky_ghi` is a good choice, it sounds the most natural when speaking it.

**Additional context**
Related: #1253

RDaxini, 2024-10-20:
All, save for just `clearsky`, seem reasonable to me. I probably prefer either `clearsky_xxx` (written as you'd say it) or `xxx_clear` (shorter)

cwhanse, 2024-10-20:
I prefer `ghi_clear` but it's a mild preference. Parallels the few terms that are a conditional irradiance, i.e., `dni_clear`, `ghi_extra`, `poa_global`, etc.

RDaxini, 2024-10-21:
I think we're set on `ghi_clear` (also `dni_clear` for where it isn't already the case)

Would updating these variables be classed as breaking changes and require some kind of deprecation? I am thinking about #2237
Is a separate PR required for each function that is changed, or is one PR for all the functions involving the changing variable appropriate? (e.g. #2273)

adriesse, 2024-11-29:
Did you consider having just one PR for these two names since the changes are so closely related?

adriesse, 2024-12-02:
> > Did you consider having just one PR for these two names since the changes are so closely related?
> 
> @adriesse I asked about PR structure [here](https://github.com/pvlib/pvlib-python/issues/2272#issuecomment-2427750656) but thought each variable had to be separate, no response so I was not really sure. I confess I think I underestimated the overlap. Is it possible to combine PRs after creation? We should do that if it would help with merging later.

Sorry I didn't speak up earlier.  I'm not the best person to advise on the PR machinations.  :(

echedey-ls, 2025-02-21:
This issue is 2/3 of being completed; it seems to me that only adding `ghi_clear` to the glossary is pending. @RDaxini are you willing to tackle it? I can do it if you prefer.

RDaxini, 2025-02-24:
> This issue is 2/3 of being completed; it seems to me that only adding `ghi_clear` to the glossary is pending. [@RDaxini](https://github.com/RDaxini) are you willing to tackle it? I can do it if you prefer.

@echedey-ls Ah yes, I'll do that. Just slipped off the radar

PR #2397 (2025-03-04): Add `ghi_clear` to glossary
 - [X] Closes #2272
 - [X] I am familiar with the [contributing guidelines](https://pvlib-python.readthedocs.io/en/latest/contributing/index.html)
 - [ ] ~Tests added~
 - [ ] ~Updates entries in [`docs/sphinx/source/reference`](https://github.com/pvlib/pvlib-python/blob/main/docs/sphinx/source/reference) for API changes.~
 - [x] Adds description and name entries in the appropriate "what's new" file in [`docs/sphinx/source/whatsnew`](https://github.com/pvlib/pvlib-python/tree/main/docs/sphinx/source/whatsnew) for all changes. Includes link to the GitHub Issue with `` [... cut]
closed 2025-03-04 by kandersolar