pvlib/pvlib-python #2279: Function `ghi_from_poa_driesse_2023` cannot have None default value for `dni_extra` optional parameter

iblasi, 2024-10-29:
**Describe the bug**
The function `ghi_from_poa_driesse_2023` defines the parameter `dni_extra` as optional and default to "None", but that causes a runtime error. `dni_extra` refers to "Extraterrestrial direct normal irradiance"
Function: https://pvlib-python.readthedocs.io/en/stable/reference/generated/pvlib.irradiance.ghi_from_poa_driesse_2023.html

**To Reproduce**
```
import pvlib
import numpy as np
import pandas as pd

# Define the site and module parameters
site_lat = 41.5	# Site latitude
site_lon = -4.0	# Site longitude
site_tz = 'Etc/UTC'	# Site timezone

# Define the tracking system parameters
axis_tilt = 0		# Tilt of the tracker axis
axis_azimuth = 180	# Tracker axis facing south
max_angle = 75		# Max tilt angle for the tracker
gcr = 0.5			# Ground coverage ratio
backtrack = True	# Enable backtracking to avoid shading

# Create the location object
location = pvlib.location.Location(site_lat, site_lon, tz=site_tz)

# Run solar position calculation
times = pd.date_range('2024-01-01 00:00', '2024-11-01 00:00', freq='15min', tz=site_tz, inclusive='left')
solar_position = location.get_solarposition(times)

# Set up the tracking system with backtracking
tracking_data = pvlib.tracking.singleaxis(
      apparent_zenith=solar_position['apparent_zenith'],
      apparent_azimuth=solar_position['azimuth'],
      axis_tilt=axis_tilt,
      axis_azimuth=axis_azimuth,
      max_angle=max_angle,
      backtrack=backtrack,
      gcr=gcr
)

# GHI 2 POA. Estimate global horizontal irradiance (GHI) from global plane-of-array (POA) irradiance
#	https://pvlib-python.readthedocs.io/en/stable/reference/generated/pvlib.irradiance.ghi_from_poa_driesse_2023.html
solar_position['poa_global'] = np.random.rand(0,800,size(len(solar_position))
solar_position['ghi'] = pvlib.irradiance.ghi_from_poa_driesse_2023(surface_tilt=tracking_data['surface_tilt'],
                                        surface_azimuth=tracking_data['surface_azimuth'],
                                        solar_zenith=solar_position['apparent_zenith'],
                                        solar_azimuth=solar_position['azimuth'],
                                        poa_global=solar_position['poa_global'])
```

**Expected behavior**
No need to define "dni_extra", or do not define it as optional

**Versions:**
 - ``pvlib.__version__``:  0.11.1
 - ``pandas.__version__``: 1.5.2
 - python: 3.10

iblasi, 2024-10-29:
I found a solution looking to another documentation example where it creates the required variable: https://pvlib-python.readthedocs.io/en/stable/gallery/irradiance-transposition/plot_rtranpose_year.html

So the issue can be solved modifying the final code in the following way to add `dni_extra` variable.
```
# GHI 2 POA. Estimate global horizontal irradiance (GHI) from global plane-of-array (POA) irradiance
#	https://pvlib-python.readthedocs.io/en/stable/reference/generated/pvlib.irradiance.ghi_from_poa_driesse_2023.html
solar_position['poa_global'] = 1000 * np.random.rand(len(solar_position))
solar_position['dni_extra'] = pvlib.irradiance.get_extra_radiation(solar_position.index)
solar_position['ghi'] = pvlib.irradiance.ghi_from_poa_driesse_2023(surface_tilt=tracking_data['surface_tilt'],
                                        surface_azimuth=tracking_data['surface_azimuth'],
                                        solar_zenith=solar_position['apparent_zenith'],
                                        solar_azimuth=solar_position['azimuth'],
                                        poa_global=solar_position['poa_global'],
                                        dni_extra=solar_position['dni_extra'])
```
**Still an issue**
Although this solves the runtime error, the documentation should be updated and do not define it as optional, or if the optional parameter `dni_extra` is None, create that value with the `get_extra_radiation` function as described.

**Note**: This example has random `poa_global` values so some NaN might appear on the `ghi` result. I suppose tht it will be due to these random values, but I haven't checked it.

cwhanse, 2024-10-29:
Confirmed. Every test of `ghi_from_poa_driesse_2023` specifies `dni_extra=1366.1`. A value of `dni_extra` is required for the private function `_ghi_from_poa` so I think setting `dni_extra=1366.1` as the default could be the fix here.

cwhanse, 2024-10-30:
> So another option would be to remove the default value here.

Requiring `dni_extra` doesn't seem like a burden on the user. As the method's originator, which would you prefer? No default, or default to the solar constant?

adriesse, 2024-10-30:
My preference is for no default value to be consistent with `perez `and` perez_driesse`.

PR #2331 (2024-12-12): Make `dni_extra` a required parameter for `ghi_from_poa_driesse_2023`
 - [x] Closes #2279
 - [x] I am familiar with the [contributing guidelines](https://pvlib-python.readthedocs.io/en/latest/contributing/index.html)
~~- [ ] Tests added~~
~~- [ ] Updates entries in [`docs/sphinx/source/reference`](https://github.com/pvlib/pvlib-python/blob/main/docs/sphinx/source/reference) for API changes.~~
 - [x] Adds description and name entries in the appropriate "what's new" file in [`docs/sphinx/source/whatsnew`](https://github.com/pvlib/pvlib-python/tree/main/docs/sphinx/source/whatsnew) for all change [... cut]
closed 2024-12-12 by kandersolar