pvlib/pvlib-python #2284: Glossary page updates tracking

RDaxini, 2024-10-30:
Discussions ongoing in #2234; I thought it would help to create a single separate issue to track suggested further updates to that page.

- [ ] Add units
- [ ] "irradiation" -> "irradiance" ([comment](https://github.com/pvlib/pvlib-python/pull/2234#discussion_r1793592373))
- [ ] Enhance definitions (refer to 'dni' example)

Supporting: #2234 
Related / additional context: #1421

RDaxini, 2024-11-25:
@kandersolar could we re-open this? still a few tasks left

kandersolar, 2024-11-25:
Looks like it was auto-closed by merging @2290.  That's caused by using [magic words](https://docs.github.com/en/issues/tracking-your-work-with-issues/using-issues/linking-a-pull-request-to-an-issue#linking-a-pull-request-to-an-issue-using-a-keyword) like "closes".  If you want to mention an issue but not close it, you have to use vaguer words.  I think "partially addresses" is safe... :P