pvlib/pvlib-python #2285: Add open-meteo iotools function

AdamRJensen, 2024-10-30:
Open-meteo offers easy access of several meteorological datasets such as ERA5.

Note, it requires a paid plan for commercial usage.

See previous discussion here:
_Originally posted by @williamhobbs in https://github.com/pvlib/pvlib-python/discussions/1484#discussioncomment-6365761_

Zhaohh0706, 2026-10-07:
@aman-coder03, you asked about this last December. Are you still working on it? If so I'll step back. Otherwise I'd like to pick it up.

Here is the scope I have in mind. Since forecast data in iotools is still an open question (#1528), I'd start with only the Historical Weather API (`/v1/archive`). That's one function, which also gives ERA5 without a CDS account:

```python
get_open_meteo_historical_weather(
    latitude, longitude, start, end,
    variables=['ghi', 'dni', 'dhi', 'temp_air', 'wind_speed'], *,
    model='era5', map_variables=True, api_key=None, timeout=60, url=None)
```

It returns `(data, meta)` with a UTC index. With `map_variables=True`, Open-Meteo names are mapped to `ghi`, `dni`, `dhi`, `bhi`, `temp_air`, `temp_dew`, `relative_humidity`, `wind_speed`/`wind_direction` (10 m, requested in m/s) and `pressure` (hPa converted to Pa). Radiation is the mean of the preceding hour, so the index marks the end of the interval, as in `get_era5`. Temperature and wind are instantaneous. I checked one day against the CDS values in `test_era5.py`, and values and timestamps agree within rounding. The docstring would state that the free API is non-commercial only, that the data are CC BY 4.0, and that `api_key` switches to the customer server. Tests would use a stored JSON response, plus a few `remote_data` tests.

Out of scope for now: the Historical Forecast, Previous Runs and Single Runs APIs, GTI, and multiple locations.

Two questions:
1. Is `model='era5'` an acceptable default, instead of the API's `best_match`, which blends in IFS runs?
2. Should the index keep end-of-interval labels like `get_era5`, or shift to mid-interval like `get_merra2`?

kandersolar, 2026-10-07:
@Zhaohh0706 this topic needs more discussion between the pvlib maintainers before any implementation.  Please do not work on it.

By the way, may I ask how pvlib came to your attention, and why you are interested in contributing to this project?  Many people have been trying to contribute (via LLM, I think) to pvlib recently, and I would like to understand why that is.

Zhaohh0706, 2026-10-07:
Yeah ofc.  

But idk what’s going on with the others; can't speak for them.( I thought trying to contribute should be a good thing? 

I picked this up because I use pvlib for clear-sky irradiance and solar position in a PV power forecasting project of mine.

The comparison with the ERA5 values in test_era5.py that I mentioned was run by opus 5.5. I have rerun it myself and verified the numbers.

Personally speaking, I studied atmospheric science and did my MSc in energy and renewable energy, which focusing on climate change mitigation, and I'm thinking about working in renewable energy forecasting. Visible contributions to the tools I use would be a bonus for me, but I'd rather leave things alone than add noise. I'm sorry if this added to the load.

cwhanse, 2026-10-07:
@Zhaohh0706 it has been noisy in here :)

Thanks for introducing yourself and letting us know you aren't someone's coding assistant spamming PRs for resume points. And thanks for your patience!

williamhobbs, 2026-10-07:
Now that we have `iotools.get_era5`, is there a need for interfacing with open-meteo for ERA5? There's the benefit of not needing an API key for open-meteo (I think), but there's also the issue of commercial use restrictions.