pvlib/pvlib-python #2288: Broken link in pvsystem.retrieve_sam

cwhanse, 2024-11-04:
The [link](https://github.com/pvlib/pvlib-python/blob/a6995c377d65ce850c2d744ad42a2a79f6d511b3/pvlib/pvsystem.py#L2036-2037) to documentation of CEC parameter sets is broken.

**Expected behavior**
Link to a current webpage? A bit of searching hasn't turned up a target for this link.

I don't think pvlib should be responsible to maintain a description of those databases. I favor just removing the dead link.

cwhanse, 2024-11-04:
@janinefreeman

kandersolar, 2024-11-05:
archive.org says it's just the PV publications page: https://web.archive.org/web/20200618135912/https://sam.nrel.gov/photovoltaic/pv-sub-page-2.html

I agree that the link is not really helpful for these databases, but for completeness, the new url is: https://sam.nrel.gov/photovoltaic/pv-publications.html

janinefreeman, 2024-11-06:
Let me know if you need any other links!

PR #2312 (2024-12-03): Remove broken link in pvsystem.retrieve_sam
 - [X] Closes #2288 
 - [X] I am familiar with the [contributing guidelines](https://pvlib-python.readthedocs.io/en/latest/contributing/index.html)
 - [X] Pull request is nearly complete and ready for detailed review.
 - [X] Maintainer: Appropriate GitHub Labels (including `remote-data`) and Milestone are assigned to the Pull Request and linked Issue.

If I understood correctly, the conclusion in #2288 is to remove the links entirely rather than update the second one with https://sam.nrel.gov/photovoltaic/pv-publications.html (?)