pvlib/pvlib-python #2295: Spectral Mismatch Calculation Integration Range

mandylewis, 2024-11-15:
As it is currently written, the `calc_spectral_mismatch_field()` function takes only the module spectral response (`sr`), the measured spectral solar irradiance (`e_sun`), and the reference spectrum (`e_ref`) as inputs. It then truncates the reference spectrum to the wavelength range of the measured spectrum to perform the SMM calculation, like so:

```math
\frac{\int^{\lambda_2}_{\lambda_1} E_{sun}(\lambda) SR(\lambda)}{\int^{\lambda_2}_{\lambda_1} E_{ref}(\lambda) SR(\lambda)} \times 
\frac{\int^{\lambda_2}_{\lambda_1} E_{ref}(\lambda) }{\int^{\lambda_2}_{\lambda_1} E_{sun}(\lambda)} 
```

where $\lambda_1$ and $\lambda_2$ are the wavelength range of the spectral irradiance measurement

I see that there was a discussion [here](https://github.com/pvlib/pvlib-python/pull/1524#issuecomment-1236131204) concerning the addition of two more optional inputs, the total irradiance of the measured spectrum (`e_sun_tot`) and the reference spectrum (`e_ref_tot`), which would change this calculation to:

```math
\frac{\int^{\lambda_2}_{\lambda_1} E_{sun}(\lambda) SR(\lambda)}{\int^{\lambda_2}_{\lambda_1} E_{ref}(\lambda) SR(\lambda)} \times 
\frac{ E_{ref,total} }{E_{sun,total}} 
```

I'm curious, why wasn't this option included in the end?

As a follow-up question, in a case where we have co-located pyranometer GHI measurements with global spectral irradiance measurements, could the GHI measured by the pyranometer be used to replace $E_{sun,total}$? Is there a strong reason why not to do this if the sensors are time-synchronized and very near each other?

I'm interested because I've found that the long-wavelength absorption of clouds is actually significant for SMM values, and truncating the spectrum in the way that `calc_spectral_mismatch_field()` currently does would underestimate the SMM for silicon modules under cloudy conditions. I was hoping to use measured pyranometer data to capture this effect instead. 

@adriesse, @RDaxini, do you have any comments?

RDaxini, 2024-11-15:
Hey Mandy, an astute observation and a good question!

> I'm curious, why wasn't this option included in the end?

I'm sure @adriesse will be able to provide a better answer as I was not part of the original implementation, but I had a scan through the discussion in #1524 and it seems to me that for the sake of simplicity it was implemented in this way with the appropriate caveat added to the docstring. I guess in (most?) situations people would be relying on spectral irradiance data alone, which typically not extend to 4000nm, in which case the truncation executed in this function would be required anyway.

> could the GHI measured by the pyranometer be used to replace esun_total?

Wait for a second opinion, but I can't think of a reason why not.

> I'm interested because I've found that the long-wavelength absorption of clouds is actually significant for SMM values, and truncating the spectrum in the way that `calc_spectral_mismatch_field()` [...]

Further developments are always welcome. If you think you could help improve the `calc_spectral_mismatch_field()` then I'd be happy to help you through the contribution process! Let's see what some others think about this.

markcampanelli, 2024-11-16:
@mandylewis The primary reference cell calibration case that I was considering in my PR review used a direct-FOV cavity radiometer for the broadband sun measurement. This thermopile is typically “more ideal” with a higher cutoff wavelength than a wide-FOV pyranometer, due to the missing glass dome in the cavity device. In any case, working from first principles is the surest guide to knowing which integration limits should be what in the formula. I’d be happy to help sketch that out in further detail.

In terms of your concurrent broadband pyranometer and spectral irradiance measurements I have secondhand knowledge that says to be aware of the time-constant difference that may exist between their response times (thermopiles can be on the order of seconds as compared to comparatively much faster solid state detectors, although some spectroradiometers scan instead of snapshotting). Spectral correction errors may be higher during highly variable conditions due to instrument lag.

mikofski, 2024-11-16:
I recommend using the equation twice to calculate two spectral mismatch value, one for the PV device and another for the pyrometer. This rewrites heading the spectral responses of both devices.
[image]
This is from: [Preliminary Investigation of
Methods for Correcting for
Variations in Solar Spectrum
under Clear Skies](https://www.nrel.gov/docs/fy10osti/47277.pdf)

markcampanelli, 2024-11-16:
To clarify my comment w.r.t. @mikofski‘s refinement of @mandylewis’s initial mismatch formula, when I spoke of a thermopile being “ideal”, this refers to both (1) its spectral response being perfectly constant (usually taken as one without loss of generality in spectral corrections where any scalar multiple typically (!) factors out in both numerator and denominator and thus cancels) AND (2) the wavelength limits of the response going from 0 to infinity. So, to the extent that one’s pyranometer is not ideal (it’s not!), then using @mikofski’s approach is indeed indicated.

Also, I cannot help but remind about the key assumption of Isc linearity w.r.t. total irradiance for any fixed spectrum.

adriesse, 2024-11-16:
> I'm curious, why wasn't this option included in the end?

It was just a question of staying focused on the use case at hand.

> As a follow-up question, in a case where we have co-located pyranometer GHI measurements with global spectral irradiance measurements, could the GHI measured by the pyranometer be used to replace E [...]

If you have the synchronized parallel measurements, I would encourage you to just try it and see what comes out!

> I'm interested because I've found that the long-wavelength absorption of clouds is actually significant for SMM values, and truncating the spectrum in the way that `calc_spectral_mismatch_field()` [...]

Don't you have Spectrafy instruments?  There would be no truncation using those.  

Your findings sound interesting.  Would you be able to propose a better assumption about the unmeasured portion of the spectrum?

markcampanelli, 2024-11-16:
Here is a spectral measurement extension technique based on how the NREL primary reference cell calibration lab extends their measured direct spectrum with a standard spectrum so that it integrates to the total irradiance measured by the direct cavity radiometer:

https://github.com/markcampanelli/pvfit/blob/main/pvfit/measurement/spectral_correction/computation.py#L201

  It should also work reasonably for global measurements as done in the technique I presented at the recent PVPMC (note overcast days here also have a notably different spectral effect): https://github.com/markcampanelli/pvpmc2024/blob/main/poster/poster-final-mark-campanelli.pdf

RDaxini, 2025-05-19:
Seems like this discussion is complete so I am closing the issue. @mandylewis hope you found the answers for which you were looking!

closed 2025-05-19 by RDaxini