pvlib/pvlib-python #2300: Add wavelength range arguments to `spectrum.average_photon_energy`

RDaxini, 2024-11-18:
**Is your feature request related to a problem? Please describe.**
The calculated APE value depends on the integration (wavelength) limits. Currently, `spectrum.average_photon_energy` performs the calculation for the full wavelength range of the input. The user is unable to calculate the APE based on a limited section of the spectrum.

**Describe the solution you'd like**
Add two new arguments to the function: `upper_wavelength` and `lower_wavelength`
Argument names are open for discussion
Default values: `None`, if value is `None` then the APE is calculated based on the full wavelength range of the input. Are there any alternative suggestions for this behaviour?

**Additional context**
I will open a PR soon for this, after wrapping up a couple of other open PRs first.

adriesse, 2024-11-18:
It's pretty easy to limit the wavelength range already by adjusting the input, isn't it?

RDaxini, 2024-11-19:
@adriesse I agree it is, but I thought it might be a more nicely packed 'complete package' if this functionality was integrated into this function? Just an idea though, I don't have to go ahead with it if it seems unnecessary:)