pvlib/pvlib-python #2304: New functionalities on pvlib/ivtools/sdm.py - different optimization techniques + polar coordinates optimization

cardenca, 2024-11-22:
Several months ago, we introduced five algorithms for computing the single-diode model (SDM) parameters using various optimization functions. Among these, we proposed a novel approach to improve the computation of SDM parameters by using polar coordinates.

Since we have already implemented these functions in Python, it might benefit the community to make them accessible through a platform like PVlib. This would allow for straightforward comparisons of different optimization methods when identifying SDM parameters from voltage/current vector pairs.

The general concept is outlined in the following link: https://cardenca.github.io/pvpmc/.

Our idea is to integrate this into pvlib/ivtools/sdm.py. However, I noticed in Issue #2252 that there is a suggestion to restructure code lines, which might make this an opportune starting point.

Questions:
1.     Would you be interested in having these methods integrated into PVlib?
2.     Could this initiative align with or complement the ideas proposed in Issue #2252 @kandersolar ? 

@cwhanse, I have finally found the time to work on this request.

markcampanelli, 2024-11-22:
@cardenca Have you run your fitting algorithms against the test IV curves here?

https://ivcurves.readthedocs.io/en/latest/

cardenca, 2024-11-22:
@markcampanelli woow, I didn't even know that page existed. I will try the algorithms there to supplement the results. Thank you!

cwhanse, 2024-11-22:
@cardenca for me the answers would be 1> Yes. and 2> not really. There's a terminology difference. In pvlib, "single diode model" refers to a set of equations that predict output given input irradiance and temperature, where "single diode equation" refers to that circuit equation that describes a single IV curve. The algorithms published in your paper appear to estimate parameters (e.g., shunt resistance) for the single diode equation, and would fit in pvlib/ivtools/sde.py rather than in the /sdm family of modules.

markcampanelli, 2024-11-22:
@cardenca I’m taking a closer look at your method. You may find it interesting that I reformulated PVfit’s SDM (equivalently, the SDE) in terms of Isc and Voc in a 2016 paper (https://ieeexplore.ieee.org/document/7329928). I actually found some numerical issues using the Voc formulation across various modeling tasks (even though it did seem promising to be able to read two model parameters right off the spec sheet). Perhaps I gave up too soon on that path (!), although I still use the model parameterization in terms of Isc instead of Iph. It is interesting to see that the polar coordinate transformation appears to make better use of the Voc parameterization!

My recent paper simplifies my 2016 approach, and is probably easier to get through. It focuses on SDM calibration, not SDE: https://www.sciencedirect.com/science/article/abs/pii/S0038092X24004079

cardenca, 2024-11-24:
@cwhanse Thank you for your reply. I will adapt the algorithms I have to include them in PVlib, specifically in pvlib/ivtools/sde.py

cardenca, 2024-11-24:
@markcampanelli Thank you very much, Mark, for your valuable insights. I will take the time to thoroughly review both of your published works (2016 and 2024). I'm sure that I can derive valuable information from them.