pvlib/pvlib-python #2309: Which conda pvlib package should be installed?

julian-belina, 2024-11-26:
Hello everyone, 

I am a bit confused by the multiple conda packages of pvlib. Should I prefer one of the packages?

1. pvlib from the pvlib channel
2. pvlib from the conda-forge channel
3. pvlib-python from the conda-forge channel

The pvlib channel is missing some of the newer versions, and the pvlib package on conda-forge is missing some of the old versions. Are any of these packages deprecated?

The documentation states that most users should install pvlib-python, but gives an example of installing pvlib from conda forge https://pvlib-python.readthedocs.io/en/stable/user_guide/installation.htm 

Is there any difference between the packages, or should they all work equally well as long as the version I want is available?

Best regards

Julian

kandersolar, 2024-11-26:
There is no meaningful difference between `pvlib` and `pvlib-python` on conda-forge.  One is essentially an alias for the other.  

The `pvlib` channel is indeed no longer updated as of [v0.9.5](https://pvlib-python.readthedocs.io/en/stable/whatsnew.html#v0-9-5-march-18-2023).  

The `pvlib` channel, and `pvlib-python` on conda-forge, are provided for backwards compatibility.  Preferring `conda-forge` and `pvlib` is recommended, but so long as you can find the version you want, there is no difference between the various options.  

It would be a welcome contribution to clarify the Installation page, in case other users encounter a similar uncertainty :)

wholmgren, 2024-11-27:
I might be overlooking something but I think we stopped advertising the `pvlib` channel as an installation option a couple of years ago. Is there current documentation that needs to be updated?