pvlib/pvlib-python #2314: Improve pvlib.tracking.singleaxis docstring

adriesse, 2024-12-02:
**Describe the bug**
The docstring does not explain what the calculated angle is based on.  At the very least it should say that it minimizes AOI and thereby maximizes DNI.

**Versions:**
 - ``pvlib.__version__``: 
 - ``pandas.__version__``: 
 - python:

**Additional context**
Add any other context about the problem here.

bandish1304, 2025-02-07:
I will be working on this issue...

kandersolar, 2025-06-17:
> if you are required to do a number of contributions 

Please also keep in mind that open-source software projects are not obligated to facilitate such assignments, and that their maintainers usually have full-time jobs elsewhere and may have very different working hours than you do :)

bandish1304, 2025-06-17:
> @kadheer<https://github.com/kadheer>
>
> This issue is not about read_ecmwf_macc, it is about pvlib.tracking.singleaxis. Where did you read about that? I haven't seen any issue open nor closed regarding that renaming.
>
> This issue is about a docstring improvement (please re-read the first comment above), you are welcome to work on it. I guess @bandish1304<https://github.com/bandish1304> has given up on this one (is that right?). And please, if you are required to do a number of contributions or this one does not match any criteria you've been tasked with, write to us in the mailing list<https://groups.google.com/g/pvlib-python?pli=1> or open a discussion with your plans/doubts.

Hi Kadheer...sorry I just saw your email right now. You can work on this issue if you like. I already close this issue from my side but you can work on this one if it's open. It should be totally fine. Thankyou

closed 2026-10-05 by kandersolar