pvlib/pvlib-python #2315: Parameters missing from ivtools.sdm.fit_desoto_sandia returns statement

RDaxini, 2024-12-03:
**Is your feature request related to a problem? Please describe.**
In addition to the parameters already listed in the returns statement in the docs of [pvlib.ivtools.sdm.fit_desoto_sandia](https://pvlib-python.readthedocs.io/en/stable/reference/generated/pvlib.pvsystem.calcparams_desoto.html#pvlib-pvsystem-calcparams-desoto), this function also returns `a_ref` and `dEgdT`, which are two parameters required for the [De Soto module performance model](https://pvlib-python.readthedocs.io/en/stable/reference/generated/pvlib.pvsystem.calcparams_desoto.html#pvlib-pvsystem-calcparams-desoto).

**Describe the solution you'd like**
Add descriptions of `a_ref` and `dEgdT` to the returns statement in the docs

**Describe alternatives you've considered**
Have I missed something here...?

kandersolar, 2024-12-03:
@cwhanse I notice a possible sign error: `fit_desoto_sandia` returns `dEgdT=0.0002677` while `calcparams_desoto` defaults to `dEgdT=-0.0002677`.  Is this a bug?

cwhanse, 2024-12-03:
> @cwhanse I notice a possible sign error: `fit_desoto_sandia` returns `dEgdT=0.0002677` while `calcparams_desoto` defaults to `dEgdT=-0.0002677`. Is this a bug?

Yes it is. New issue or fix here?
 
> If so, it may affect the matlab version as well: https://github.com/sandialabs/MATLAB_PV_LIB/blob/master/pvl_desoto_parameter_estimation.m#L276

Technically, no, since the Matlab function doesn't return dEgdT.

kandersolar, 2024-12-03:
Opened #2321 so we can close this one with #2317.

PR #2317 (2024-12-03): Add missing parameters to returns statement in ivtools.sdm.fit_desoto_sandia
 - [X] Closes #2315
 - [X] I am familiar with the [contributing guidelines](https://pvlib-python.readthedocs.io/en/latest/contributing/index.html)
 - [X] Tests added
 - [ ] Adds description and name entries in the appropriate "what's new" file in [`docs/sphinx/source/whatsnew`](https://github.com/pvlib/pvlib-python/tree/main/docs/sphinx/source/whatsnew) for all changes. Includes link to the GitHub Issue with `` :issue:`num` `` or this Pull Request with `` :pull:`num` ``. Includes contributor name and/or GitHub usernam [... cut]
closed 2024-12-03 by kandersolar