pvlib/pvlib-python #2318: v0.11.2 release planning

kandersolar, 2024-12-03:
The v0.11.2 release is scheduled for Dec 15, but let's shoot for end of next week (Dec 13).

Please tag any missing PRs/issues for v0.11.2, and take a look through the [v0.11.2 milestone](https://github.com/pvlib/pvlib-python/milestone/48) to see what should be pushed to v0.11.3.  We have a bit of a review backlog built up.

RDaxini, 2024-12-03:
#2310 and #2311 might need to be pushed back to 11.3. Will update as my schedule evolves

RDaxini, 2024-12-03:
#2184 -> 11.3?

kandersolar, 2024-12-13:
FYI I'm planning to release v0.11.2 in a few hours.  If any unmerged PRs are ready to go, please point them out :)

RDaxini, 2024-12-13:
#2306 will have to move to 11.3 as I can't finish it in the next couple of hours. I could get it done tonight (MST) if we want it merged with 11.2 though.

kandersolar, 2024-12-13:
I was about to say that there's no rush and we can push #2306 to the next release, but it would be nice for #2306 and #2274 to be done in the same release.  Let's plan for releasing 0.11.2 on Monday Dec 16th then.

PR #2335 (2024-12-17): Finalize v0.11.2
 - [x] Closes #2318
 - [x] I am familiar with the [contributing guidelines](https://pvlib-python.readthedocs.io/en/latest/contributing/index.html)
 - ~[ ] Tests added~
 - ~[ ] Updates entries in [`docs/sphinx/source/reference`](https://github.com/pvlib/pvlib-python/blob/main/docs/sphinx/source/reference) for API changes.~
 - [x] Adds description and name entries in the appropriate "what's new" file in [`docs/sphinx/source/whatsnew`](https://github.com/pvlib/pvlib-python/tree/main/docs/sphinx/source/whatsnew) for all changes. Includes link to the GitHub Issue with `` :issue:`nu [... cut]
closed 2024-12-17 by kandersolar