pvlib/pvlib-python #2321: `pvlib.ivtools.sdm.fit_desoto_sandia` has incorrect sign for returned `dEgdT`

kandersolar, 2024-12-03:
`fit_desoto_sandia` incorrectly returns `+0.0002677` instead of `-0.0002677` for `dEgdT`:

https://github.com/pvlib/pvlib-python/blob/891778cc9ceb11e159991d08a6d6cb2d09661bdf/pvlib/ivtools/sdm.py#L874-L886

That value is used both for the returned value and in the equations that compute other SDM parameters.  I've not inspected the math to see whether the sign should be fixed in both places, or only for the returned value.

Noticed in #2315.

cwhanse, 2024-12-03:
The bug stems from a laziness in the original report, where I dropped a pair of parathenses. The Matlab code was transcribed from a printed equation, and the pvlib code was transcribed from Matlab.

The default value from De Soto et al. paper is `dEgdT=-0.0002577`, and the relevant equation in the De Soto model is

```
E_g = EgRef * (1 + dEgdT*(Tcell_K - Tref_K))
```

The code in `fit_desoto_sandia` could have been rendered

```
 dEgdT = -0.0002677 
 x_for_io = const['q'] / const['k'] * ( 
     1. / tok - (1. / tck[u] + dEgdT * (tc[u] - const['T0']) / tck[u]))
```
 
I think this should be the fix, rather than adding a minus sign to the later line of code.

echedey-ls, 2024-12-03:
May I tackle this issue?

cwhanse, 2024-12-03:
@echedey-ls please do. LMK if you run into issues with tests, I can help with the known solutions.

echedey-ls, 2024-12-05:
> The default value from De Soto et al. paper is `dEgdT=-0.0002577`

@cwhanse , just in case, that 5 is a typo, right? It should be 0.0002677.

cwhanse, 2024-12-05:
> > The default value from De Soto et al. paper is `dEgdT=-0.0002577`
> 
> @cwhanse , just in case, that 5 is a typo, right? It should be 0.0002677.

That is correct.

PR #2322 (2024-12-13): Add minus sign in front of dEgdT, `sdm.py`

 - [x] Closes #2321
 - [x] I am familiar with the [contributing guidelines](https://pvlib-python.readthedocs.io/en/latest/contributing/index.html)
 - [x] Tests added
 - ~[ ] Updates entries in [`docs/sphinx/source/reference`](https://github.com/pvlib/pvlib-python/blob/main/docs/sphinx/source/reference) for API changes.~
 - [x] Adds description and name entries in the appropriate "what's new" file in [`docs/sphinx/source/whatsnew`](https://github.com/pvlib/pvlib-python/tree/main/docs/sphinx/source/whatsnew) for all changes. Includes link to the GitHub [... cut]
closed 2024-12-13 by kandersolar