pvlib/pvlib-python #2323: Incorrect refraction elevation cutoff in ephemeris solar position algorithm

AdamRJensen, 2024-12-04:
The refraction correction used in the ephemeris solar position algorithm has a cutoff at -1 degree, which is not mentioned in the [original publication](https://www.osti.gov/servlets/purl/6377969) or the provided Fortran code:

https://github.com/pvlib/pvlib-python/blob/6d886dc0fa8e0fa1d011e9332f609cee526c9bbd/pvlib/solarposition.py#L839

To be consistent with the publication I suggest that we remove the lower limit and apply refraction correction to all negative solar elevation angles.

[image]

AdamRJensen, 2024-12-04:
ICYMI @IoannisSifnaios @kandersolar

cwhanse, 2024-12-04:
The reference publication is here: https://pvpmc.sandia.gov/app/uploads/sites/243/2022/10/[Engineering-Astronomy.pdf](https://pvpmc.sandia.gov/app/uploads/sites/243/2022/10/Engineering-Astronomy.pdf)

It doesn't mention correction below -0.575 degrees, that I can find.

My guess is that the -1 degree limit was added to the Matlab code because we didn't think refraction correction was meaningful when the sun's disc was below the horizon.

I have no objection to removing the limit, or adding a comment pointing back to the Matlab code. I also have no objection to removing the `ephemeris` function as long as there is an equally fast solar position algorithm in the library.

kandersolar, 2024-12-04:
+1 to removing `ephemeris` eventually, and leaving it as-is in the meantime.  But good to document the issue regardless.

A bit of context: @AdamRJensen, @IoannisSifnaios, and I have a project underway comparing solar position algorithms.  We plan to contribute implementations of one or more algorithms to pvlib-python, depending on the results of the comparison.  We expect some of them to be better than `ephemeris` in all respects (speed, accuracy, period of validity, clear reference).

closed 2026-09-29 by AdamRJensen