pvlib/pvlib-python #2329: Model comparison tables

kandersolar, 2024-12-10:
SciPy's [documentation](https://docs.scipy.org/doc/scipy/reference/optimize.html#scalar-functions) has this nice table comparing root finding methods:

[image]

We should consider including similar tables in cases where pvlib provides semi-interchangeable alternatives, for example models for transposition, decomposition, and module temperature.  Spectral mismatch has been previously suggested ([comment](https://github.com/pvlib/pvlib-python/issues/2065#issuecomment-2133455259)).  The structure/contents of the comparisons tables would depend on the models being compared, but they could (for example) compare:

- transposition: necessary "extra" inputs (airmass, dni_extra, ...) and what POA components are output (horizon, circumsolar, isotropic)
- spectral mismatch: necessary inputs and which cell technologies are directly supported

Where would the tables go in the documentation?  They could go at the top of the corresponding API reference page, e.g. [here](https://pvlib-python.readthedocs.io/en/stable/reference/irradiance/transposition.html), but I'd rather see them incorporated into new topic-focused User Guide pages along the lines of #1754.

adriesse, 2024-12-10:
Seems like a good idea to me.  I would look for it under "Users Guide" if I knew it existed.

RDaxini, 2024-12-11:
I also think the user guide is the right place. The new subsections created through #2302 would support this.

Could I work on one of these? I'd be happy to do the spectral mismatch one. If there are no takers for the transposition models one, I wouldn't mind working on that afterwards.

echedey-ls, 2024-12-11:
I agree with the already mentioned inclusion in the user guide, but for ease of accessing it, the newly created page / section can also be cross-referenced in the submodule page (maybe in some admonition?).