pvlib/pvlib-python #2330: Add new PVsyst temperature model parameters

cwhanse, 2024-12-11:
PVsyst has added a set of temperature model parameters for roof mounted systems that allow air flow beneath the modules, termed "semi-integration".

**Describe the solution you'd like**
Add this parameter set to pvlib\temperature\pvsyst_cell, and as a value for PVSystem's `racking_model` parameter.

#2293 should be merged first.

Muhammad-Rebaal, 2025-03-10:
Hi @cwhanse !
I would like to solve this issue and do the required enhancement so could you please assign this issue to me?
The solution I'd like to propose is : 

`PVsyst's "semi-integration" model to pvlib.temperature.pvsyst_cell and as a racking_model option in PVSystem for improved accuracy in roof-mounted system modeling.`

Thank you!

cwhanse, 2025-03-10:
@Muhammad-Rebaal sure thing.

Details of the PVsyst model are [here](https://www.pvsyst.com/help-pvsyst7/thermal_loss.htm).

closed 2025-06-03 by kandersolar