pvlib/pvlib-python #2342: Replace pytz with standard library's zoneinfo

markcampanelli, 2024-12-23:
**Is your feature request related to a problem? Please describe.**

I find `pytz` to be somewhat confusing and cumbersome. Since Python 3.9, `pytz` can been replaced by the standard library's [zoneinfo](https://docs.python.org/3/library/zoneinfo.html) package for full and up-to-date IANA timezone support. Furthermore, it appears that localization of "naive" `timedate` timestamps is fully supported without using `pytz`. For example:

```python
import datetime
from pprint import pprint
import zoneinfo

pprint(zoneinfo.available_timezones())
time_naive = datetime.datetime(1974, 6, 22, 18, 30, 15)
print(time_naive)
time_aware = time_naive.replace(tzinfo=zoneinfo.ZoneInfo("Etc/GMT+5"))
print(time_aware)
print(time_aware.astimezone(zoneinfo.ZoneInfo("UTC")))
```

In particular, the `pvlib.location.Location` class can be streamlined significantly by using only the `zoneinfo.ZoneInfo` class as the timezone attribute. Of course, `pytz` would still be lurking in the background due to its significant usage in `pandas`.

**Describe the solution you'd like**
Switch from `pytz` to `zoneinfo`. We could optionally deprecate `pytz` usage initially in the `pvlib.location.Location` constructor, but my sense is that simply removing it entirely will not break many people who typically instead use timezone strings in the initializer. Also, https://pvlib-python.readthedocs.io/en/latest/user_guide/timetimezones.html would need a significant update.

**Describe alternatives you've considered**
 Keep using `pytz`.

**Additional context**
See https://github.com/pvlib/pvlib-python/pull/2339 for an example of `pytz`'s full removal.

kandersolar, 2024-12-23:
I confess this is the first I've heard of zoneinfo, but it seems like the community at large is agreed that switching from pytz to zoneinfo is a good idea.  Thanks @markcampanelli for bringing it up!

I'm seeing some people saying that Windows doesn't include a tz database by default.  Depending on `tzdata` may be necessary: https://docs.python.org/3/library/zoneinfo.html

> Of course, pytz would still be lurking in the background due to its significant usage in pandas.

Good point, although it looks like pandas 3.0 will make pytz an optional dependency (https://github.com/pandas-dev/pandas/pull/59089).

markcampanelli, 2024-12-24:
@kandersolar Did you happen to discover if there was a more explicit way to "localize" a datetime using `ZoneInfo` other than the following?
```python
time_naive = datetime.datetime(1974, 6, 22, 18, 30, 15)
time_aware = time_naive.replace(tzinfo=zoneinfo.ZoneInfo("Etc/GMT+5"))
```

markcampanelli, 2025-01-10:
- 

> I'm seeing some people saying that Windows doesn't include a tz database by default. Depending on `tzdata` may be necessary: https://docs.python.org/3/library/zoneinfo.html

It seems like this is only with significantly out of date Windows installations. All the Windows tests added in #2341 that use `zoneinfo.ZoneInfo` pass.

markcampanelli, 2025-03-12:
UPDATE: #2341 updated the "internal" time-zone representation in location objects, i.e., `pvlib.location.Location._zoneinfo`, to use the `zoneinfo` library. Remaining work seems to be mostly deprecating/removing the `pytz` field and updating the documentation away from `pytz` throughout but especially in https://pvlib-python.readthedocs.io/en/stable/user_guide/timetimezones.html.

closed 2026-08-15 by markcampanelli