pvlib/pvlib-python #2350: can't install pvfactors

laestis, 2025-01-07:
Hi,

When I am trying to install pvfactors I have the following error:

    AttributeError: module 'pkgutil' has no attribute 'ImpImporter'. Did you mean: 'zipimporter'?

From what I understood pkgutil has been removed from pip.

my pip version is pip 24.0 from /usr/lib/python3/dist-packages/pip (python 3.12)

thanks

joseph-sch, 2025-01-28:
I've opened a PR in [https://github.com/pvlib/solarfactors](solarfactors), https://github.com/pvlib/solarfactors/pull/22, to upgrade the code for compatibility with Shapely 2.0+ and therefore Python 3.12. All tests pass and the PR is ready for review.

kandersolar, 2025-02-07:
A new version of solarfactors has been released with the fixes in @joseph-sch's PR.  It should now install without issue on Python 3.12 and above.  If you encounter any problems, please open an issue at https://github.com/pvlib/solarfactors/

closed 2025-02-07 by kandersolar