pvlib/pvlib-python #2355: pv.clearsky.detect_clearsky breaks easily with optimizer, not caused by window size/ num of data points

doitslow, 2025-01-14:
Simply input the following measured and clearsky values of ghi to pvlib.clearsky.detect_clearsky would cause scaling factor optimizer to break.
[958.0, 961.0, 961.0, 962.0, 963.0, 963.0, 965.0, 966.0, 966.0, 967.0, 967.0, 969.0, 969.0, 970.0, 970.0, 970.0, 971.0, 972.0, 973.0, 973.0, 974.0, 974.0, 976.0, 976.0, 977.0, 977.0, 978.0, 979.0, 979.0, 980.0]
[1118.4577383713079, 1119.40369782888, 1120.3339376894762, 1121.2484402052023, 1122.1471879333149, 1123.0301637264158, 1123.8973513157873, 1124.7487329945209, 1125.5842931054435, 1126.404015710825, 1127.2078851703661, 1127.9958861475639, 1128.7680036084419, 1129.5242228205852, 1130.2645293588396, 1130.9889090973, 1131.6973482150947, 1132.3898336543418, 1133.066351272327, 1133.7268886329964, 1134.3714331314945, 1134.999972469113, 1135.6124946523566, 1136.2089879919956, 1136.7894411076372, 1137.3538429214848, 1137.9021826627754, 1138.4344498670578, 1138.950634716054, 1139.4507266679402]

[image]

My guess is inital run finds zero clearsky periods?

doitslow, 2025-01-14:
Is it necessary to input whole day's data such that optimizer has a higher chance of finding clearsky periods to avoid this error?

markcampanelli, 2025-01-14:
@doitslow In order to allow others to try to reproduce, can you provide the version of pvlib that you are using, as well as the exact call that is failing?

doitslow, 2025-01-14:
Sure. I am using version==0.11.0.

doitslow, 2025-01-14:
Or just about any random time series data input gives the same error.

doitslow, 2025-01-14:
Indeed, it seems to be a version issue. I tried version==0.11.2 on my server with no issue. Thanks!

doitslow, 2025-01-14:
@markcampanelli BTW, is there any recommendation about how long should the input time series be? I am working on Folsom dataset with 3 years of measurements, how should I segment the whole dataset to get the most accurate clearsky detection results? Say, segment by day, by month, by year or even just input 3 years' data in total?

markcampanelli, 2025-01-14:
@doitslow I must defer your question to someone more qualified on the algorithm than me.

cwhanse, 2025-01-14:
@doitslow indeed, with pvlib<0.11.2, when _no_ clearsky data are found, the optimizer fails and exits with that error. In v0.11.2 the optimization algorithm was improved, it is good to know it is robust in this situation.

No clearsky data are found because the data (the array around 980 W/m2) generally differ from the clearsky model data (around 1120) by more than `max_diff=75`, (W/m2), the threshold used for the mean and maximum of each series over each window. Longer time series won't fix a systematic mismatch between data and clearsky model.

You can try a different clearsky model that more closely matches your data, or increase `max_diff` to allow some periods to be detected. No promises, but my guess is that if some periods are labelled as clearsky, the optimizer will move the clearsky model towards the data (that's what the `alpha` factor does).

Any objections to closing this as an Issue? I don't see something to fix in the code.

doitslow, 2025-01-15:
@cwhanse Thank you for the detailed reply, it is very helpful.

closed 2025-01-15 by doitslow