pvlib/pvlib-python #2360: Requesting feedback on Sandia’s PV Performance Modeling Idea No. 1: Inverter modeling

mtheristis, 2025-01-23:
As part of the Sandia’s PV Performance Modeling program (https://pvpmc.sandia.gov/), we are sharing our model development ideas to seek feedback based on relevance, importance, and potential impact. 

Your input will help us prioritize our efforts for fiscal years 2025-2027. Any model developments from our PV Performance Modeling program will be validated, documented in peer-reviewed articles, and made available in pvlib-python. If you are already working on a similar idea, please reach out—we would love to collaborate!

Idea No. 1.	Improve inverter modeling: Extend inverter models to account for common operating conditions and constraints: e.g., off-unity power factor, MPPT voltage range, DC current limit, curtailment due to temperature. The goal is to account for inverter operating states that affect output power by > 1%.

@kandersolar @cwhanse @jsstein @leliadeville

adriesse, 2025-01-23:
> If you are already working on a similar idea, please reach out—we would love to collaborate!

I am working on this topic—I would love to collaborate!

williamhobbs, 2025-01-23:
I think some work here would definitely be valuable. Some models, like temperature derates, should be pretty easy to implement basic useful versions of. 

MPPT voltage ranges are important and I don't think get enough attention (I ignored them for a long time...). And I'm not sure that performance engineers have good examples to work from when they need to set expectations for plants that operate off of unity power factor, so that could be helpful.

kurt-rhee, 2025-01-24:
This would certainly be useful to my company.  I started working on this but then had to move to a different topic.  Glad to hear that @adriesse is working on something!

Peque, 2025-01-27:
This would be very valuable I think, and would address common and widespread use cases (PV has great adoption for self-consumption use cases on roofs or ground-mounted).

grgfgh, 2025-01-29:
it would be interesting for the company I work for. even more if the models are tested with real data