pvlib/pvlib-python #2363: Requesting feedback on Sandia’s PV Performance Modeling Idea No. 4: PV module modeling

mtheristis, 2025-01-23:
As part of the Sandia’s PV Performance Modeling program (https://pvpmc.sandia.gov/), we are sharing our model development ideas to seek feedback based on relevance, importance, and potential impact.

Your input will help us prioritize our efforts for fiscal years 2025-2027. Any model developments from our PV Performance Modeling program will be validated, documented in peer-reviewed articles, and made available in pvlib-python. If you are already working on a similar idea, please reach out—we would love to collaborate!

**Idea No 4. PV module modeling**. Assess the extent to which existing models accurately predict the PV behavior of “newer” technologies, e.g., half-cut cells, TOPCon. For example, modeling the effects of shading on half-cut cells, or ensuring that existing models can predict TOPCon performance accurately. 

@kandersolar @cwhanse @jsstein @leliadeville

adriesse, 2025-01-24:
I think on this topic Sandia’s PV Performance Modeling program could really help the modelling community by publishing open lab and/or field data for the "newer" technologies.

Peque, 2025-01-27:
Would be great to see further studies on partial shading affections.

At [ieco.io](https://ieco.io) we have started working on ways to simplify calculations for these use cases (i.e.: find a good balance between precision and computational efficiency for complex non-uniform affections like the ones that can occur in an idustrial rooftop for example). We are going to make documentation, code and results open for anybody to use for any purpose and to try to help push forward open models that can be verified, studied and improved. We will not focus on the physical model (i.e.: not about trying to create a model to fit on-field data), but rather study the ways in which simplifications/approximations can impact results when compared to a baseline case (considered as the "reference" when it comes to precision).