pvlib/pvlib-python #2365: Requesting feedback on Sandia’s PV Performance Modeling Idea No. 6: Improve pvlib's default clear-sky model

mtheristis, 2025-01-23:
As part of the Sandia’s PV Performance Modeling program (https://pvpmc.sandia.gov/), we are sharing our model development ideas to seek feedback based on relevance, importance, and potential impact.

Your input will help us prioritize our efforts for fiscal years 2025-2027. Any model developments from our PV Performance Modeling program will be validated, documented in peer-reviewed articles, and made available in pvlib-python. If you are already working on a similar idea, please reach out—we would love to collaborate!

**Idea No 6. Improve pvlib's default clear-sky model.** Depending on location, predicted annual insolation shows up to 18% difference from more sophisticated clear-sky models. Advanced clear-sky models are complex and require specialized inputs, making them only practical for expert institutions.  The idea here is to develop a simple-to-use surrogate model that approximates the output of advanced models and compare to other simple clear-sky models using SURFRAD and BSRN data, to examine whether a model trained on Americas data only can be generalized to the rest of the world.

@kandersolar @cwhanse @jsstein @leliadeville

williamhobbs, 2025-01-23:
I've found that some of my work is very sensitive to the clear sky model that I use. A specific example is when normalizing power and forecasts to clear sky for forecasting related work/analysis. 

This work seems very interesting and worth doing.

adriesse, 2025-01-24:
> Improve pvlib's default clear-sky model

*Which model is the default?*

To get more accurate clear sky values you need to feed in more data about the atmosphere.  This is true even for the simple models already in pvlib, and these conditions can change a lot from one day to the next.

The proposal here seems to be to avoid using more data about the atmosphere, and to focus on reducing annual bias.  This seems realistic, but you would still need data about regional differences.  

A realistic first option would be to use accurate sources of clear sky data to create suitable maps of representative linke turbidity, precipitable water, and aod700 for the existing models, and maybe tune them a bit.  The spatial and temporal resolution would be up for discussion.

Having said that, I still think it would be nice to have some more advanced clear sky models in pvlib that **do** use more inputs.

I am working on related things—I would love to collaborate!