pvlib/pvlib-python #2366: Intro tutorial: different results for Procedural and Object oriented examples

cwhanse, 2025-01-23:
The Procedural and Object-oriented examples use different atmospheric pressure, thus the AC energy results are slightly different.

E.g.,

Procedural
```
Tucson           445963.743286
```

Object-oriented
```
Tucson           445945.594522
```

The individual results are not wrong, but it may confuse new users to get different numbers from two simulations that could be assumed to be the same.

The difference is introduced by this line in the Procedural section:
```
    solpos = pvlib.solarposition.get_solarposition(
        time=weather.index,
        latitude=latitude,
        longitude=longitude,
        altitude=altitude,
        temperature=weather["temp_air"],
        pressure=pvlib.atmosphere.alt2pres(altitude),
    )
```

The Object-oriented section simply passes `weather` to `mc.run_model`, which finds `'pressure'` in the `weather` file and uses it.

To make the two examples produce identical numbers, change the Procedural section to
```
    solpos = pvlib.solarposition.get_solarposition(
        time=weather.index,
        latitude=latitude,
        longitude=longitude,
        altitude=altitude,
        temperature=weather["temp_air"],
        pressure=weather["pressure"],
    )
```

manojks1999, 2025-01-26:
/assign

echedey-ls, 2025-01-26:
> /assign

We don't have that kind of commands integrated here, but go ahead!

You may find https://pvlib-python.readthedocs.io/en/stable/contributing/index.html useful. Do not hesitate to ask for anything :)

cwhanse, 2025-01-26:
assigned to @manojks1999 😄

manojks1999, 2025-01-27:
@cwhanse , made a PR for the changes

PR #2367 (2025-01-30): Fix for the different results for Procedural and Object oriented exam…
…ples

 - [ * ] Closes #2366
 - [ * ] I am familiar with the [contributing guidelines](https://pvlib-python.readthedocs.io/en/latest/contributing/index.html)
 - ~~[ ] Tests added~~
 - [ * ] Pull request is nearly complete and ready for detailed review.
 - [x] Maintainer: Appropriate GitHub Labels (including `remote-data`) and Milestone are assigned to the Pull Request and linked Issue.
closed 2025-01-30 by cwhanse