pvlib/pvlib-python #2368: Suggestion: reorganise user guide folder with subfolders

RDaxini, 2025-01-29:
**Is your feature request related to a problem? Please describe.**
The [user guide folder ](https://github.com/pvlib/pvlib-python/tree/main/docs/sphinx/source/user_guide) is becoming pretty packed with a range of different files. More files are soon to be added, e.g. through #2353. 

**Describe the solution you'd like**
Would it be helpful to organise the files using subfolders? These could align with the structure of the new readthedocs subsections (#2310), as one possibility.
I think subfolders would make navigating the folder easier now, and even more so once more pages are added (#2329).

**Describe alternatives you've considered**
No subfolders...

cwhanse, 2025-01-29:
We're having this discussion at my lab. In my view, the current User Guide would be better separated into three subheadings:

1 Getting Started - the Introductory tutorial is the only content we have that I would put here
2 Modeling Guide - steps through the modeling process giving a bit of theory for each step, and a guide to implementation in pvlib
3 Technical Reference - material that is supplementary to the modeling guide, such as the details about the single diode equation and its solution

RDaxini, 2025-01-30:
@cwhanse just to clarify, are you talking about subsections for the section navigation pane on the user-facing readthedocs site, or sorting the `.rst` files, which are inside the user guide folder, into subfolders of those names?

cwhanse, 2025-01-30:
I was thinking about the user-facing navigation panes.

RDaxini, 2025-01-30:
Ah okay, since I thought that matter was settled in #2310, in this issue I was thinking about the file organisation. I'm open to reworking the user-facing subsections though. Is it possible to re-open a merged PR?

cwhanse, 2025-01-30:
Let's not go back on merged PRs. What I put up here as an idea, is a subdivision of the Advanced usage section from #2310, into Modeling Guide and Technical Reference.

Sorry I confused your proposal (sort the files) with the user-facing menus.

RDaxini, 2025-03-14:
Is documentation the appropriate label?

PR #2411 (2025-04-10): Rearrange user guide files into subfolders
 - [X] Closes #2368
 - [X] I am familiar with the [contributing guidelines](https://pvlib-python.readthedocs.io/en/latest/contributing/index.html)
 - [ ] Adds description and name entries in the appropriate "what's new" file in [`docs/sphinx/source/whatsnew`](https://github.com/pvlib/pvlib-python/tree/main/docs/sphinx/source/whatsnew) for all changes. Includes link to the GitHub Issue with `` :issue:`num` `` or this Pull Request with `` :pull:`num` ``. Includes contributor name and/or GitHub username (link with `` :ghuser:`user` ``).
 - [X] New code i [... cut]
closed 2025-04-10 by kandersolar