pvlib/pvlib-python #2377: zenith vs solar_zenith in psm3 variable maps

williamhobbs, 2025-02-06:
Should `'zenith': 'solar_zenith_angle', ` in `REQUEST_VARIABLE_MAP` instead be `'solar_zenith': 'solar_zenith_angle', `? In `VARIABLE_MAP` it is `solar_zenith`.

https://github.com/pvlib/pvlib-python/blob/2ca40aa377d0871905f3c12f7ab0de0d437e2e21/pvlib/iotools/psm3.py#L45-L61

cwhanse, 2025-02-06:
I think @williamhobbs is correct, line 52 should be 'solar_zenith': 'solar_zenith_angle'

ayushjariyal, 2025-02-07:
Is this issue still unresolved?

ayushjariyal, 2025-02-07:
@williamhobbs Is this issue still unresolved?

williamhobbs, 2025-02-07:
> Is this issue still unresolved?

@ayushjariyal Yes, it it still unresolved. I already have another PR open, and will probably not start on this until that one is resolved, so feel free to work on it yourself, if that's why you are asking.

ayushjariyal, 2025-02-13:
@williamhobbs Can you please review the changes and let me know if any further improvements are needed?

williamhobbs, 2025-02-13:
> [@williamhobbs](https://github.com/williamhobbs) Can you please review the changes and let me know if any further improvements are needed?

@ayushjariyal, looks good to me!

PR #2381 (2025-02-21): Replace zenith with solar_zenith in variable map for pvlib:psm3

 - [ ] Closes #2377
closed 2025-02-21 by kandersolar