pvlib/pvlib-python #2383: loss_townsend: Update Townsend Snow Model References

kjsauer, 2025-02-08:
In regards to this paragraph in the function "loss_townsend":

"""
    The parameter `string_factor` is an enhancement added to the model after
    publication of [1]_ **per private communication with the model's author**. The
    definition for snow events documented above is **also based on private
    communication with the model's author**.
"""

Please replace the first author private communication reference (re string_factor) with a reference to this Townsend poster:
https://www.nrel.gov/docs/fy25osti/90585.pdf

Please replace the second author private communication reference (re snow events definition) with a reference to this Townsend poster:
[10.13140/RG.2.2.14299.68647](http://dx.doi.org/10.13140/RG.2.2.14299.68647)

Thanks!

mikofski, 2025-02-08:
Thanks Ken!

ayushjariyal, 2025-02-09:
Is this issue still unresolved?

cwhanse, 2025-02-09:
@kjsauer are you sure about the string_factor reference? I went through that NREL file and didn't find a poster with Townsend as an author.

kjsauer, 2025-02-09:
@cwhanse: I got the year wrong. It should be 2023, not 2024. Apologies. Please try this link:
https://www.nrel.gov/docs/fy24osti/87918.pdf

kandersolar, 2025-02-11:
I am not sure our code is wholly consistent with the SPI reference.  We define `snow_events` as:

```
    snow_events : array-like
        Number of snowfall events each month. Snow events are defined as days
        in the month that have snowfall greater than 1 inch. May be int or
        float type for the average events in a typical month. Referred to as N
        in [1]_.
```

Whereas the poster defines it as `# of monthly snow events of at least 1" per event`, which makes no mention of days being a relevant unit.  Couldn't a long snow event last multiple days, or oppositely one day could have several short events within it?

kjsauer, 2025-02-12:
It was simplest to clarify this point at the source. Please see: doi.org/10.13140/RG.2.2.14299.68647

PR #2384 (2025-02-12): Updating  Townsends snow model references

 - [ ] Closes #2383