pvlib/pvlib-python #2385: Outdated NSRDB PSM3 documentation link returns 404 Not Found

ayushjariyal, 2025-02-10:
**Describe the bug**
The documentation link for PSM3 Data Download in the pvlib code and references is outdated and returns a 404 Not Found error.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to the following NSRDB API documentation link:
`https://developer.nrel.gov/docs/solar/nsrdb/psm3_data_download/`
2. The page returns 404 Not Found.

**Expected behavior**
The link should point to the correct documentation page for PSM3 Data Download in the NSRDB API.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Versions:**
 - ``pvlib.__version__``: 
 - ``pandas.__version__``: 
 - python:

**Additional context**
1. The API endpoints seem to be working, but the documentation link is no longer valid.
2. The latest documentation link should be updated in the pvlib codebase and references.
3. Possible alternative documentation: [NSRDB API Docs](`https://developer.nrel.gov/docs/solar/nsrdb/`)

RDaxini, 2025-02-11:
Could you link the docs page where you found this broken link?
References on [iotools.get_psm3](https://pvlib-python.readthedocs.io/en/stable/reference/generated/pvlib.iotools.get_psm3.html) seem to be working fine. I noticed the link in the changed in version 10.0 warning on that page does not work, but that's different to the link you have shared.

ayushjariyal, 2025-02-11:
Thanks for looking into this! The broken link is located in the file :

📂 pvlib/iotools/psm3.py

Specifically, it appears in the "changed in version 10.0" warning within that file. While the references in iotools.get_psm3 seem to be working fine, this particular link still points to the outdated page that returns a 404 error.

williamhobbs, 2025-02-11:
There's an issue with the link here:

https://github.com/pvlib/pvlib-python/blob/2ca40aa377d0871905f3c12f7ab0de0d437e2e21/pvlib/iotools/psm3.py#L1-L4

There's also the confusing note `Get PSM3 TMY`, which relates to the point @kandersolar brought up here https://github.com/pvlib/pvlib-python/issues/2326#issuecomment-2643842599 about having different API endpoints (TMY, etc.) in the same function. 

It should probably list out the different documentation links if the psm3 tools are going to stay in a single function, e.g.,

```python
"""
Get PSM3 data from several endpoints. See:
https://developer.nrel.gov/docs/solar/nsrdb/psm3-2-2-tmy-download/
https://developer.nrel.gov/docs/solar/nsrdb/psm3-2-2-download/
https://developer.nrel.gov/docs/solar/nsrdb/psm3-5min-download/
"""
```

kandersolar, 2025-02-11:
`git grep` is a handy tool for finding things like this.  It shows that @williamhobbs found the only location of that particular url:

```bash
$ git grep -n psm3_data_download
pvlib/iotools/psm3.py:3:see https://developer.nrel.gov/docs/solar/nsrdb/psm3_data_download/
```