pvlib/pvlib-python #2386: Estimation the effective nominal power in a PV generator under outdoor condition

JRAnguloPUCP, 2025-02-11:
The estimation of the nominal power of a photovoltaic generator is crucial for evaluating the operational state of PV plants. International standards such as IEC 61829 and ASTM E2848-13 propose methods to estimate the nominal power under clear sky conditions. However, these standards impose strict requirements:

1. They require stable meteorological conditions, including high irradiance levels (>800 W/m²) and low wind speeds.
2. They exclude data from partially cloudy days, limiting their applicability in regions with frequent cloud cover.
3. They require specialized equipment such as I-V curve tracers, which are impractical for large PV plants.

This becomes a problem when **Estimating effective nominal Power**:

-  The nominal power of a PV system is expected to match the manufacturer’s nominal power,  under the IEC and ASTM standards, it is impossible to assess this characterization on partially cloudy days in tropical areas.
- Contractual Agreements in PV Plant Transactions. For instance, the purchase and sale of solar power plants, investors require an accurate estimation of the actual nominal power to ensure compliance with expected performance guarantees.
- If only clear sky days can be used for evaluation, many locations (e.g., tropical regions) would lack reliable data. The IEC and ASTM methodologies may underestimate power losses in degraded PV systems.

We propose a new methodology that allows **estimating the effective nominal power even under non-ideal conditions.** [This methodology](https://doi.org/10.1016/j.solener.2021.12.015), validated in previous research studies, is based on non-parametric statistical filtering and presents several advantages:

**Mathematical Approach**

- Instead of discarding data from partially cloudy days, we use a Kernel Density Estimation (KDE) method to extract the most probable nominal power value from daily measurements.
- The methodology filters out anomalies caused by shading, inverter saturation, and ambient changes.

**Robust Estimation**

- The method was successfully tested on a 109.44 kW PV plant in Granada, Spain.
- It achieved a power estimation uncertainty of less than 1%, comparable to the IEC and ASTM standards under ideal conditions.
- Application in Diverse Climates
- The methodology was further tested in challenging environments, such as a desert region (Lima, Peru) and a tropical region (Chachapoyas, Peru), paper draft.
- Unlike standard approaches, it remained reliable under cloudy conditions.

**Application in Contractual and Degradation Analysis**

- The method provides a more accurate assessment of a PV plant’s current performance.
- It can be used to verify performance guarantees in solar power transactions.
- It offers a more flexible alternative to traditional standards, making it applicable to more locations worldwide.

We have analyzed existing methodologies, including:

[IEC 61829](https://cdn.standards.iteh.ai/samples/15535/4157371cb8674bcd8a9d0365154c0767/IEC-61829-2015.pdf) (I-V Curve Translation to STC)

- Requires specialized equipment to measure the I-V curve of each string.
- Difficult to implement in large-scale PV plants.

[ASTM E2848-13](https://www.astm.org/e2848-13r18.html)

- Requires at least 3 clear sky days with stable irradiance (around 1000 W/m²).
- Not applicable in partially cloudy environments.

[Martínez-Moreno et al. (2012)]( https://doi.org/10.1002/pip.1102) Method

- Uses a linear regression approach for nominal power estimation.
- Excludes data from non-ideal conditions, limiting its applicability.

Compared to these alternatives, this procedure:

- [ ] Extends applicability to partially cloudy days.
- [ ] Alternative to the need for I-V curve tracers. 
- [ ] Uses robust statistical techniques to ensure accuracy.

The procedure requires monitoring data from the photovoltaic plant under real operating conditions. The necessary parameters are:

**1. Irradiance at the module plane (W/m²)**. 

- Measured using a pyranometer or a calibrated photovoltaic module.
- A sampling frequency of at least every 30 seconds or 1 minute is required.

**2. Photovoltaic module temperature (°C)**

- Measured using a thermocouple, PT100 sensor, or through indirect methods (e.g., open-circuit voltage).

**3. DC output power (W)**

- Measured at the maximum power point (MPP) of the inverter.

**4. Reference data from the module datasheet**

- Manufacturer's nominal power under Standard Test Conditions (STC).
- Power temperature coefficient (%/°C).
- Reference irradiance at STC (1000 W/m²).

The output provides **Actual Effective Nominal Power (W or kW)** 

This methodology is currently validating its applicability in various climates and real-world conditions. **Would the pvlib-python community be interested in integrating this method as an enhancement to the existing nominal power estimation tools?**

mikofski, 2025-02-11:
Is this an announcement, discussion or feature request? It might be better to post this in the [pvlib Google group](https://groups.google.com/g/pvlib-python) or convert to GH discussion?

cwhanse, 2025-02-11:
@JRAnguloPUCP thanks for this well-written post. I'm going to move this to a Discussion, rather than an Issue, since we're not at the point of adding specific features.

My personal view is that software supporting capacity testing isn't a good fit for pvlib-python, but may be a good fit for pvlib/pvanalytics. pvanalytics doesn't have anything like that currently, but it's a young project and could embrace that addition to scope. 

The only comparable software I know of is the pvcaptest library, which is specific to the ASTM 2848 method. There is currently great interest in capacity test methods in the PV industry in the US. It makes sense to share software to allow comparison among alternative methods.