pvlib/pvlib-python #2390: Reference for Gueymard 1993 air mass equation incorrect

mandylewis, 2025-02-12:
I believe that reference [4] on this page: https://pvlib-python.readthedocs.io/en/v0.7.1/generated/pvlib.atmosphere.get_relative_airmass.html

Should actually be: 

C. Gueymard, “Development and performance assessment of a clear sky spectral radiation model,” in Proceedings 22nd ASES Conference, Solar ’93, 1993, pp. 433–438

The equation is 3b in this reference.

cwhanse, 2025-02-12:
@mandylewis the equation for the Gueymard93 air mass model appears as Eq. 22 in [4]. Do you think the Solar '93 reference adds value? Or we could add a comment in the code that points to [4], Eq. 22

mandylewis, 2025-02-13:
Thanks for pointing me in the right direction. At first glance, Equation 22 appeared different from the equation implemented in the source code:

```
    elif 'gueymard1993' == model:
        am = (1.0 / (np.cos(zenith_rad) +
              0.00176759*(z)*((94.37515 - z) ** - 1.21563)))
```

But now I see that they're identical, just that Eq. 22 uses elevation angle while Eq. 3b in the other reference uses zenith angle.

Since the actual implementation in the code matches the reference I suggested, maybe it would be worth adding a note somewhere?  Or maybe just adding the comment pointing to equation 22 would be enough.

cwhanse, 2025-02-13:
A comment in the code pointing to Eq. 22 would be welcome.

I don't want to replace the Solar Energy reference [4] with the Solar '93 reference, because ASES proceedings are sometimes hard to locate. We could add the Solar '93 reference; I'm OK with that.

AdamRJensen, 2025-02-22:
I reached out to Gueymard about whether there was potentially an even better paper to cite, and he responded with the following comment:
> That is actually an antiquated formula.
> The correct/current one is Eq. B-8 from doi:[10.1016/S0038-092X(03)00195-6](https://doi.org/10.1016/S0038-092X(03)00195-6), with different coefficients.

[image]

cwhanse, 2025-02-22:
That 2003 reference would be a new airmass model, no objections from me to add it as `gueymard2003`, or `smarts` if we can verify that's what SMARTS uses.

kandersolar, 2025-02-22:
We already have that one:

https://github.com/pvlib/pvlib-python/blob/171f10ca1c2a73ea41a06e4c6ecf4de1f53523e7/pvlib/atmosphere.py#L245-L247

AdamRJensen, 2025-02-23:
Good catch!

Then I suggest that we just add the additional reference to the 1993 model that Mandy listed.

PR #2424 (2025-04-04): Add reference for Gueymard 93 relative airmass model
 - [x] Closes #2390
 - [x] I am familiar with the [contributing guidelines](https://pvlib-python.readthedocs.io/en/latest/contributing/index.html)
 - ~~[ ] Tests added~~
 - ~~[ ] Updates entries in [`docs/sphinx/source/reference`](https://github.com/pvlib/pvlib-python/blob/main/docs/sphinx/source/reference) for API changes.~~
 - [x] Adds description and name entries in the appropriate "what's new" file in [`docs/sphinx/source/whatsnew`](https://github.com/pvlib/pvlib-python/tree/main/docs/sphinx/source/whatsnew) for all changes. Includes lin [... cut]
closed 2025-04-04 by kandersolar