pvlib/pvlib-python #2394: `scipy.interpolate.interp1d` is discouraged

kandersolar, 2025-02-20:
SciPy now discourages the use of `interp1d`.  From https://docs.scipy.org/doc/scipy-1.15.2/reference/generated/scipy.interpolate.interp1d.html :

> This class is considered legacy and will no longer receive updates. While we currently have no plans to remove it, we recommend that new code uses more modern alternatives instead. For a guide to the intended replacements for [interp1d](https://docs.scipy.org/doc/scipy-1.15.2/reference/generated/scipy.interpolate.interp1d.html#scipy.interpolate.interp1d) see [1-D interpolation](https://docs.scipy.org/doc/scipy-1.15.2/tutorial/interpolate/1D.html#tutorial-interpolate-1dsection).

We use `interp1d` in a few places:

```bash
$ git grep -n interp1d
docs/examples/shading/plot_partial_module_shading_simple.py:41:from scipy.interpolate import interp1d
docs/examples/shading/plot_partial_module_shading_simple.py:181:    """convenience wrapper around scipy.interpolate.interp1d"""
docs/examples/shading/plot_partial_module_shading_simple.py:182:    f_interp = interp1d(np.flipud(df['i']), np.flipud(df['v']), kind='linear',
pvlib/iam.py:443:        See scipy.interpolate.interp1d for more options.
pvlib/iam.py:473:    from scipy.interpolate import interp1d
pvlib/iam.py:486:    interpolator = interp1d(theta_ref, iam_ref, kind=method,
pvlib/spectrum/response.py:9:from scipy.interpolate import interp1d
pvlib/spectrum/response.py:70:    interpolator = interp1d(SR_DATA[0], SR_DATA[1],
```

I suppose there is no rush to change, but updating these to use whatever scipy recommends at that link might be a good first PR for some pvlib user interested in becoming a contributor.

prady0t, 2025-03-01:
I'll be glad to work on this!

MihirGore23, 2025-03-02:
.take

MihirGore23, 2025-03-02:
I updated the interpolation code to use `make_interp_spline` instead of `interp1d` for better accuracy and stability. I also made sure the extrapolation and boundary conditions work properly across all input ranges. Kindly review in Pull Request #2401