pvlib/pvlib-python #2403: Consider switching to `src` layout

kandersolar, 2025-03-05:
> > To my mind, src is a separate discussion, and doesn't need to be done in https://github.com/pvlib/pvlib-python/pull/2277, if at all.
> 
> I brought it up because it seems to me that the majority of projects that use `/tests` also use `/src`. Indeed most of the linked packages on the PyOpenSci reference use `/pkg/tests` rather than `/pkg` + `/tests`. Also `pytest` docs [include examples](https://docs.pytest.org/en/stable/explanation/goodpractices.html#choosing-a-test-layout) for either `/src` + `/tests` OR `/pkg/tests` but not `/pkg` + `/tests`. Maybe it's ok but I like to follow the more popular patterns unless I know why I am deviating from them. 

 _Originally posted by @wholmgren in [#2271](https://github.com/pvlib/pvlib-python/issues/2271#issuecomment-2669668889)_

#2277 moved the package from a flat layout to a "halfway-src" layout.  We should consider adopting the src layout entirely, as the python community seems broadly agreed that the src layout is beneficial.

echedey-ls, 2025-03-13:
> as the python community seems broadly agreed that the src layout is beneficial

(Relevant source): https://www.pyopensci.org/python-package-guide/package-structure-code/python-package-structure.html#the-src-layout-and-testing

mikofski, 2025-03-13:
Idk, [the definitive source](https://poquitopicante.blogspot.com/2015/04/recommended-python-project-layout.html) recommends `core` or `lib` :laughing: