pvlib/pvlib-python #2406: spectral model for BIPV in vertical facades

jesuspolo, 2025-03-11:
pvlib includes several models for estimating the spectral mismatch including different technologies, that are basically designed for conventional PV. They depend on the parameters that mostly affects to the incoming solar spectrum (airs mass as the major influencer, and precipitable water and AOD in some moe recent models. The case of BIPV in building facades is particularly interesting since the major influence on the spectrum, in addition to precipitable water and AOD, relies in the air mass and angle of incidence. This influence is not contemplated in the usual models and thus the application to BIPV is limited. 

We have developed recently a new model that includes the dependence on the air mass, angle of incidence, AOD and precipitable water. The model was developed convering all possible facade orientarions (surface azimuth), by computing thousands of solar spectra with SMARTS 2.9. It considers also for technologes, m Si, a Si, CdTe and CIGS. The details are in this recent publication (DOI: 10.1016/j.renene.2025.122820)

https://www.sciencedirect.com/science/article/pii/S0960148125004823?via%3Dihub

The model is very straightforwad and it includes the potential effect of broadband albedo, through a correction to the main model that was developed for a standard albedo of 0.2. 

If this proposal is of interest I could implement the corresponding function in python in a similar way of the spectral models already implemented.

Thanks in advance for the comments and suggestions

J Polo

jesuspolo, 2025-03-11:
In theory I think that it is more general; the fact to my understanding is that for vertical tilt (maybe also for generally very high tilt angles) the angle of incidence determines somehow the diffuse or beam contribution to the spectral POA. I have only analysed the vertical case, due to the objective of BIPV facade application; and in this case I found a very good correlation when I included both the airmass and the angle of incidence. In addition, I did not explore the impact of highly reflective environment and focused only on broadband albedoes corresponding to some common materials in urban environment (aspahl, concrete, brick...)

cwhanse, 2025-03-11:
I see. Including AOI as a predictor is accounting for some of the different spectral content of direct and diffuse irradiance.

cwhanse, 2025-03-11:
Is the SMM predicted by this model meant to multiply the POA irradiance before accounting for reflections, or after reducing POA irradiance for reflections?

jesuspolo, 2025-03-11:
Good point!; the model has two contributions: the SMM as a result of massive spectral calculations with SMARTS2.9 assuming 0.2 of albedo, and a secondary albedo correction that resulted from a sensitivity calculation of spectra for a few cases of broadband albedo. In my opinion, in the first part of the model, the SMM meant to multiply the POA before accounting for reflections, since this accounts for the air mass, precipitable water, AOD and the AOI influence. The reflections would depend, in my opinion, on the specific surroundings, and maybe generalize this part would require more research.

closed 2025-12-16 by kandersolar